ARMmbed/mbed-os #6417: Remove References to Yotta, module.json, and other OS3 stuff

loverdeg-ep, 2018-03-21:
### Description
- Type: Enhancement
- Related issue: [Handbook-#462](https://github.com/ARMmbed/Handbook/issues/462#issuecomment-374846529)
- Priority: Minor

## Enhancement

When perusing the mbed OS source files I often come across preprocessor references to yotta, which as far as I know, is no longer used in any capacity for mbed development. Though a minor distraction and usually found in FEATURE_STORAGE, it does nonetheless cause me to double take.

There also seems to be some module.json files laying around as well this seems to have caused confusion on at least 2 occasions in myself and [Handbook-#462](https://github.com/ARMmbed/Handbook/issues/462#issuecomment-374846529)

**Suggested enhancement**

Does it make sense to remove some of these references?

**Pros**

Feel free to fill this in for me

**Cons**

Feel free to fill this in for me

loverdeg-ep, 2018-03-23:
@theotherjimmy  Bumping this since I noticed you were doing some spring cleaning.

Though I'm starting to feel like Yotta might still be in use.

theotherjimmy, 2018-03-23:
Ah, well some of these are in parts of the code base that I don't controll:
```
./features/FEATURE_BLE/targets/TARGET_NORDIC/TARGET_MCU_NRF51822/sdk/module.json
./features/FEATURE_BLE/targets/TARGET_NORDIC/TARGET_MCU_NRF51822/module.json
./features/FEATURE_BLE/targets/TARGET_NORDIC/TARGET_NRF5/module.json
./features/FEATURE_BLE/module.json
./features/FEATURE_COMMON_PAL/mbed-client-randlib/module.json
./features/FEATURE_COMMON_PAL/mbed-trace/module.json
./features/FEATURE_COMMON_PAL/nanostack-libservice/module.json
./features/FEATURE_COMMON_PAL/sal-stack-nanostack-eventloop/module.json
./features/FEATURE_COMMON_PAL/mbed-coap/module.json
./features/nanostack/FEATURE_NANOSTACK/mbed-mesh-api/module.json
```

@SeppoTakalo @pan- It looks like there are some modules in your departments.

bridadan, 2019-02-13:
Just updating here as we've closed this internally (and this will be closed shortly). I've  done quite a few updates to our code base to remove mentions of yotta and Mbed OS 3 stuff (see above referenced PRs).

The other teams that handle `FEATURE_STORAGE` and the like have been notified and they have open tasks to remove the references. So work will still continue on this even though this particular issue is being closed.

Thanks very much for filing this issue @loverdeg-ep, there's still much to improve but definitely a step in the right direction!

closed 2019-02-14 by ciarmcom