ARMmbed/mbed-os #6420: MbedCRC

pilotak, 2018-03-21:
Hi,
I just discovered that CRC library is baked into OS, i run the example which was fine now a want to implement CRC8_MAXIM here is how i do it:
```cpp
#include "mbed.h"
int main() {
    MbedCRC<0x31, 8> ct(0, 0, false, false); // MAXIM

    char  test[] = "123456789";
    uint32_t crc = 0;

    printf("\nPolynomial = 0x%lx  Width = %d \n", ct.get_polynomial(), ct.get_width());

    ct.compute((void *)test, strlen((const char*)test), &crc);

    printf("The CRC of data \"123456789\" is : 0x%lx\n", crc);
    return 0;
}
```
unfortunately it doesn't compile, is this class not meant to be used for other but listed/supported CRC?
```
BUILD/NUCLEO_F429ZI/GCC_ARM/main.o: In function `main':
C:\ARM\test/.\main.cpp:43: undefined reference to `mbed::MbedCRC<49ul, (unsigned char)8>::MbedCRC(unsigned long, unsigned long, bool, bool)'
collect2.exe: error: ld returned 1 exit status
[ERROR] BUILD/NUCLEO_F429ZI/GCC_ARM/main.o: In function `main':
C:\ARM\test/.\main.cpp:43: undefined reference to `mbed::MbedCRC<49ul, (unsigned char)8>::MbedCRC(unsigned long, unsigned long, bool, bool)'
collect2.exe: error: ld returned 1 exit status

[mbed] ERROR: "c:\users\pavel\python\python.exe" returned error code 1.
[mbed] ERROR: Command "c:\python\python.exe -u C:\ARM\test\mbed-os\tools\make.py -t GCC_ARM -m NUCLEO_F429ZI --source . --build .\BUILD\NUCLEO_F429ZI\GCC_ARM" in "C:\ARM\test"
```

### Description
- Type: Bug  | Question
- Priority:  Minor

**Target**
STM32F429

**Toolchain:**
GCC_ARM

0xc0170, 2018-03-24:
@deepikabhavnani Please review

deepikabhavnani, 2018-03-24:
https://github.com/ARMmbed/mbed-os/pull/6442 will fix the issue. Default constructor for template class was suppose to be in header file and not CPP file.

closed 2018-04-04 by deepikabhavnani