ARMmbed/mbed-os #6421: SerialWireOutput class outside of mbed namespace

AGlass0fMilk, 2018-03-21:
https://github.com/ARMmbed/mbed-os/blob/58f57954caca95d30908a07c225d0fe227d95ae4/drivers/SerialWireOutput.h#L22

I'm trying to develop with the SerialWireOutput class to do some debug printing. It appears that this class isn't declared inside the mbed namespace, contrary to all(?) the other drivers.

I'm trying to eliminate use of just including "mbed.h" to limit code size. This driver is causing compilation issues and I think it's because it's not declared part of the mbed namespace.

Any thoughts @marcuschangarm?

marcuschangarm, 2018-03-22:
SerialWireOutput should probably be in the mbed namespace, but would that solve your problem? I would be surprised if it worked without including mbed.h!

What code size increase do you see when you include mbed.h? It really shouldn't affect code size and if it does we should fix it.

AGlass0fMilk, 2018-03-22:
There's been discussion in several issues, notably #5679, about the issue of including "mbed.h". I've already seen conflicts from user namespace pollution. And for the sake of clarity and control I would rather just include the modules I need to use.

It becomes very easy to treat mbed as a black box when you just include mbed.h, and then you can run into problems that have no apparent cause.

marcuschangarm, 2018-03-22:
Cool! Do you mind submitting that change in a PR please? Thanks!

AGlass0fMilk, 2018-03-22:
Sure thing, see PR #6427

closed 2018-04-04 by cmonr