ARMmbed/mbed-os #6436: GR-LYCHEE - mbed-os-example-blinky fails to compile

yennster, 2018-03-22:
## Bug

**Target**
GR-LYCHEE

**Toolchain:**
Online compiler

**Expected behavior**
mbed-os-example-blinky compiles successfully and .bin file downloads for GR-LYCHEE target.

**Actual behavior**
Tons of warnings and the following errors:
```
Error: "/tmp/b4IrdO", line 9: Error:  #5: cannot open source input file "mem_RZ_A1LU.h": No such file or directory
Error:   #include "mem_RZ_A1LU.h"
Error:                           ^
Error: /tmp/b4IrdO: 0 warnings, 1 error
Error: Pre-processor step failed for '/build/.link_script.sct'
Error: Image needs at least one load region.
Error: Not enough information to list the image map.
Info: Unable to download. Fix the reported errors...
```

**Steps to reproduce**
Follow this link to import mbed-os-example-blinky into the online compiler for the GR-LYCHEE board: https://os.mbed.com/compiler/#platform:Renesas-GR-LYCHEE and check the mbed-os-example-blinky dropdown and "Update this program and libraries to latest revision" then click OK, then click Compile.

kegilbert, 2018-03-22:
Reproduced on my end as well through the online compiler with a fresh import of the blinky example. 

However GCC_ARM and ARM (ARM Compiler 5.06 update 6 (build 750)) both compiled correctly through mbed-cli (mbed-os-5.7, 16bac101a6b7b4724023dcf86ece1548e3a23cbf).

I'd need the online compiler output files in that tmp directory to figure out what's trying to pull that file in.

yennster, 2018-03-22:
When compiling with GCC_ARM thru Mbed CLI:

```
Unknown object name found in GCC map file: linker stubs
Unknown object name found in GCC map file: LONG 0x1800f438 __etext
Unknown object name found in GCC map file: LONG 0x20020000 __data_start__
Unknown object name found in GCC map file: LONG 0xa7c (__data_end__ - __data_start__)
Unknown object name found in GCC map file: LONG 0x1800feb4 __etext2
Unknown object name found in GCC map file: LONG 0x20200000 __nc_data_start
Unknown object name found in GCC map file: LONG 0x0 (__nc_data_end - __nc_data_start)
Unknown object name found in GCC map file: LONG 0x20020a80 __bss_start__
Unknown object name found in GCC map file: LONG 0x2e20 (__bss_end__ - __bss_start__)
Unknown object name found in GCC map file: LONG 0x20200000 __nc_bss_start
Unknown object name found in GCC map file: LONG 0x0 (__nc_bss_end - __nc_bss_start)
+------------------+-------+-------+-------+
| Module           | .text | .data |  .bss |
+------------------+-------+-------+-------+
| [fill]           |  1064 |     4 |    24 |
| [lib]/c.a        | 16803 |  2472 |    89 |
| [lib]/gcc.a      |  2664 |     0 |     0 |
| [lib]/misc       |   224 |    12 |    28 |
| [misc]           |    56 |     0 |     0 |
| main.o           |   104 |     4 |    24 |
| mbed-os/cmsis    |  1304 |     0 |  4084 |
| mbed-os/drivers  |   216 |     4 |   716 |
| mbed-os/hal      |  1636 |     4 |    64 |
| mbed-os/platform |  1846 |     4 |   270 |
| mbed-os/rtos     | 13590 |   174 |  5988 |
| mbed-os/targets  |  6581 |     6 |   521 |
| Subtotals        | 46088 |  2684 | 11808 |
+------------------+-------+-------+-------+
Total Static RAM memory (data + bss): 14492 bytes
Total Flash memory (text + data): 48772 bytes
```

theotherjimmy, 2018-03-23:
I already have a PR up to fix this. It's 8 days old at this point, and should be live in the next week or so.

theotherjimmy, 2018-03-23:
> I'd need the online compiler output files in that tmp directory to figure out what's trying to pull that file in.

It's the linker script.

theotherjimmy, 2018-03-23:
> Do you know of a good contact on the online compiler team who's familiar with the linking process? 

No such person exists. I'm the person that maintains the online build tools.

theotherjimmy, 2018-03-23:
> pointed out the directory structure is a bit unique for the Renese boards and may be causing an issue.

It's actually the include in the linker file that's causing problems.

TomoYamanaka, 2018-05-23:
@yanesca  Could you close this issue because this was already resolved?

closed 2018-05-23 by 0xc0170