ARMmbed/mbed-os #6438: Stream: add jlink rtt to log

woodsking2, 2018-03-23:
Type: Enhancement
Priority: Minor

Reason:
    Some chip(like nrf52832) has only one uart used for communication。

Suggested enhancement
    A rtt class from stream used for log

Pros:
    can log without uart

Cons:
     need a jlink

seppestas, 2018-03-23:
IMO it makes more sense to develop this as a library instead of adding it to mbed-os.

0xc0170, 2018-03-23:
@woodsking2 Have you had a look at `SerialWireOutput` class?

I recall there was a port for rtt before filehandle improvements landed. Then it was reworked and should be possible to find a reference in the PR, I might add it here once I find it.

0xc0170, 2018-03-24:
cc @kjbracey-arm

woodsking2, 2018-03-27:
accept it
https://github.com/ARMmbed/mbed-os/issues/6438#issuecomment-375622048
Let's make it be a library

closed 2018-03-27 by woodsking2

kjbracey-arm, 2018-09-20:
Was struggling to find the aforementioned rtt port today  - adding a link for others:

https://github.com/0x6d61726b/mbed-os-retarget-segger-rtt

geoffrey-vl, 2020-05-25:
That seems to have disappeared, mirror: https://github.com/krekeltronics/mbed-os-retarget-segger-rtt