ARMmbed/mbed-os #6444: crash_log_parser crashes

bmcdonnell-ionx, 2018-03-24:
I'm troubleshooting an intermittent "imprecise" HardFault. `crash_log_parser` crashed when trying to parse a crash log from firmware where I had disabled the write buffer use during default memory map access with the instruction `SCnSCB->ACTLR &= ~(SCnSCB_ACTLR_DISDEFWBUF_Msk);`.

References:
https://community.nxp.com/docs/DOC-103810
http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0553a/CHDCBHEE.html

### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
LPC4088

**Toolchain:**
GCC_ARM

**Toolchain version:**
MCUXpresso IDE v10.1.1 [Build 606] [2018-01-02] 
(and whatever version of GCC_ARM that comes with)

**mbed-cli version:** 
1.5.0 (with https://github.com/ARMmbed/mbed-cli/pull/642)

**mbed-os sha:**
`ad284b280 (tag: mbed-os-5.8.0-rc2, tag: mbed-os-5.8.0, arm/mbed-os-5.8, mbed-os-5.8) Merge pull request #6352 from ARMmbed/release-candidate`

**DAPLink version:**

**Expected behavior**

[crash_log_parser](https://github.com/ARMmbed/mbed-os/tree/master/tools/debug_tools/crash_log_parser) successfully parses a crash log to produce crash info.

**Actual behavior**

```
me@pc ~/Documents/proj/troubleshooting
$ ./crash_log_parser.py crash-log.txt fw-build-dbg.axf fw-build-dbg.map

Crash Info:
Traceback (most recent call last):
  File "./crash_log_parser.py", line 194, in <module>
    main(args.crashlog, elfhelper)
  File "./crash_log_parser.py", line 164, in main
    print("\tCrash location = %s [0x%s] (based on PC value)" % (pc_name.strip(), str(pc_val)))
UnboundLocalError: local variable 'pc_name' referenced before assignment
```

**Steps to reproduce (maybe)**

- Write a program that crashes with a HardFault on an ARM Cortex-M4. (Might need to be "imprecise"; IDK. Cortex-M3 would probably work.)
- Include the instruction `SCnSCB->ACTLR &= ~(SCnSCB_ACTLR_DISDEFWBUF_Msk);` to force precise faults.
- Export to MCUXpresso IDE. Build. Load and run firmware.
- Make/watch it crash. Capture the `MbedOS Fault Handler` output, and run the `crash_log_parser` on it.
- Observe failure to parse as above. Maybe.

0xc0170, 2018-03-24:
@SenRamakri Please review

cc @ARMmbed/team-nxp

mmahadevan108, 2018-03-26:
@bmcdonnell-ionx Thank you for finding this. Is this a fix to the DAPLink HICHAL driver?

Could you submit a patch.

bmcdonnell-ionx, 2018-03-26:
I did have some crashes later which it could parse, so I'm not sure if you'll be able to reproduce this. Maybe you can give it a shot, and if not, then we wait and see.

(Sorry I can't share the files with you to troubleshoot. It's application firmware.)

@mmahadevan108 

> Could you submit a patch.

I would have if I knew how to fix the issue.

> Is this a fix to the DAPLink HICHAL driver?

I don't know where the problem lies.

bmcdonnell-ionx, 2018-03-26:
I'm using an Embedded Artists' [LPC4088 QuickStart Board](https://www.embeddedartists.com/products/boards/lpc4088_qsb.php).

SenRamakri, 2018-03-26:
As far as I see this is an issue in the tool itself, nothing to do with DAPLink driver. I'll look into this.

SenRamakri, 2018-03-26:
@bmcdonnell-ionx - Do you have the crash dump which caused the tool to fail?

bmcdonnell-ionx, 2018-03-26:
@SenRamakri

> Do you have the crash dump which caused the tool to fail?

I'm glad you asked...

On further inspection, I found that the crash dump file I was using used CR (`\r`) for linebreaks, whereas another that was fine used CRLF (`\r\n`). I'm not sure how that happened; maybe I captured one from a PuTTY terminal window, and the other from a log file, or something.

Can you make the parser more tolerant?

[crash-log-01--CR.txt](https://github.com/ARMmbed/mbed-os/files/1848680/crash-log-01--CR.txt)
[crash-log-01--CRLF.txt](https://github.com/ARMmbed/mbed-os/files/1848681/crash-log-01--CRLF.txt)

SenRamakri, 2018-03-26:
@bmcdonnell-ionx - Thanks for sharing the info and feedback. I will look into making the tools more tolerant. Eventually, we may integrate this script/feature into other mbed debug tools.

SenRamakri, 2018-06-27:
Fix in process - https://github.com/ARMmbed/mbed-os/pull/7356

SenRamakri, 2018-07-02:
Fix has been merged.

closed 2018-07-02 by SenRamakri