ARMmbed/mbed-os #6449: mbed deploy fails randomly with Mbed OS 5.8

teetak01, 2018-03-26:
### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
All

**Toolchain:**
Any

**mbed-cli version:** 
(`mbed --version`)

**mbed-os sha:**
5.8.0-RC2

**Expected behavior**
mbed deploy works always

**Actual behavior**
00:00:14.231 + mbed deploy --protocol ssh
00:01:00.404 error: Your local changes to the following files would be overwritten by checkout:
00:01:00.404 	targets/TARGET_Realtek/TARGET_AMEBA/TARGET_RTL8195A/serial_api.c
00:01:00.404 Please, commit your changes or stash them before you can switch branches.
00:01:00.404 Aborting
00:01:00.406 [mbed] ERROR: Unable to update "mbed-os" to rev #ad284b28064b

This happens highly randomly in CI.

0xc0170, 2018-03-26:
Does this happen with earlier mbed OS versions?

@theotherjimmy @screamerbg

teetak01, 2018-03-26:
Nope, only with 5.8.0

0xc0170, 2018-03-26:
It seems the file is one to blame, not certain why we havent seen this in our CI . @studavekar Please review

```
git diff
warning: CRLF will be replaced by LF in targets/TARGET_Realtek/TARGET_AMEBA/TARGET_RTL8195A/serial_api.c.
The file will have its original line endings in your working directory.
diff --git a/targets/TARGET_Realtek/TARGET_AMEBA/TARGET_RTL8195A/serial_api.c b/targets/TARGET_Realtek/TARGET_AMEBA/TARGET_RTL8195A/serial_api.c
index 5f1cccb6f..46f2dde5f 100644
--- a/targets/TARGET_Realtek/TARGET_AMEBA/TARGET_RTL8195A/serial_api.c
+++ b/targets/TARGET_Realtek/TARGET_AMEBA/TARGET_RTL8195A/serial_api.c
@@ -327,7 +327,7 @@ void serial_irq_set(serial_t *obj, SerialIrq irq, uint32_t enable)
         if(enable) {
             if (irq == RxIrq) {
                 log_uart_irq_set(&stdio_uart_log, IIR_RX_RDY, enable);
-                serial_log_irq_en |= SERIAL_RX_IRQ_EN;
+                serial_log_irq_en |= SERIAL_RX_IRQ_EN;
```

cc @geky @bulislaw

0xc0170, 2018-03-26:
Steps to reproduce locally with blinky

```
λ mbed import git@github.com:ARMmbed/mbed-os-example-blinky.git
[mbed] Importing program "mbed-os-example-blinky.git" from "git@github.com:ARMmbed/mbed-os-example-blinky.git" at latest revision in the current branch
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #16bac101a6b7

λ cd mbed-os-example-blinky.git\mbed-os\

\mbed-os-example-blinky.git\mbed-os (mbed-os-5.7 -> origin)
λ git st
On branch mbed-os-5.7
Your branch is up to date with 'origin/mbed-os-5.7'.

nothing to commit, working tree clean

\mbed-os-example-blinky.git\mbed-os (mbed-os-5.7 -> origin)
λ git co master
Switched to branch 'master'
Your branch is up to date with 'origin/master'.

\mbed-os-example-blinky.git\mbed-os (master -> origin)
λ git st
On branch master
Your branch is up to date with 'origin/master'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

        modified:   targets/TARGET_Realtek/TARGET_AMEBA/TARGET_RTL8195A/serial_api.c

no changes added to commit (use "git add" and/or "git commit -a")
```

0xc0170, 2018-03-26:
I hope that this is now resolved, and further actions will be taken for file line endings

From our coding style guide: `Unix line endings are default option for files.`

@OPpuolitaival @SeppoTakalo @kjbracey-arm - Should we review .gitattributes for line endings for this repository? also astyle can handle the check.

teetak01, 2018-03-27:
Thanks, issue has been fixed in #6450

closed 2018-03-27 by teetak01

kjbracey-arm, 2018-04-03:
I think the .gitattributes in the repo is fine.

The problem is .gitattributes is only effective if obeyed by a Git client - so it works fine if committing using normal command-line git with default settings. But if people are doing anything else it can be bypassed - eg if someone uploads a file via the Github web UI, or overrides Git settings locally.

So would be a good idea to incorporate into astyle checks or similar, so there is a server-end check.

Not sure how this particular one got through.

0xc0170, 2018-04-03:
> So would be a good idea to incorporate into astyle checks or similar, so there is a server-end check.

@OPpuolitaival Can you please look at this, can add this check to CI ?
We had really hard time because of such as "silly" file change. Worth adding as server-end check 👍

kjbracey-arm, 2018-04-03:
@OPpuolitaival - See also my more detailed comments on testing on #6450.