ARMmbed/mbed-os #6451: Add option to read out if serial transmission is done

marcemmers, 2018-03-26:
### Description
- Type: Enhancement
- Priority: Minor

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**
When running in a low power setup we are experiencing issues with serial transmission. When we transmit one or multiple characters using for example puts of putc the transmission will loop until all characters are put into the tx buffer of the peripheral. This however doesn't check if the last char has been transmitted.

When the device goes to deepsleep before the last char has been transmitted the peripheral will halt because the clock disappears, causing corrupt data transfers.

**Suggested enhancement**
I want to add an option to the drivers to be able to read out the transmission status, just like readable and writable. Then it will be up to the user if they want to use the function after a transmission or if they don't care.

**Pros**
Add function to get some more information from the serial peripheral.
Solves issue for very low power use.

**Cons**
None to my knowledge other than implementation time for all processors.

0xc0170, 2018-03-26:
cc @bulislaw @c1728p9

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTHAL-236*

closed 2019-10-31 by ciarmcom