ARMmbed/mbed-os #6454: Consecutive ticker callbacks result more timer interval interrupt requests than expected

gaborkertesz, 2018-03-26:
- Type: Bug
- Priority: Major

**Target**
ARM_CM3DS_MPS2 

**Toolchain:**
GCC_ARM|ARM|IAR

**mbed-os sha:**
458e726b1e2c32fb2ea6238771190d6fdd78e18d

**Expected behavior**
In the example two 1sec callbacks are requested and 120 interval timestamps in us_ticker_set_interrupt are collected. In best case only in every other second would be a new interval request, which means 120 samples would be collected after 120 seconds.
Chained callbacks would be great to trigger 1 interrupt instead of small delay and 2 interrupts.

**Actual behavior**
After 28seconds the 120 samples are collected, the first 20 attached.
For every registered callback there are two interval requests and there is a small interval request between the requested callbacks (check 2->3 or 4->5)
index   usec interval
[0]    1879048232
[1]    1021242
[2]    1021242
[3]    1021387
[4]    1021387
[5] 	2021242
[6] 	2021242
[7] 	2021387
[8]  	2021387
[9] 	3021242
[10]	3021242
[11]	3021387
[12]	3021387
[13]	4021242
[14]	4021242
[15]	4021387
[16]	4021387
[17]	5021242
[18]	5021242
[19]	5021387
[20]	5021387

**Steps to reproduce**

Example of main.cpp
#include "mbed.h"

Ticker ticker1;
Ticker ticker2;

#define NR_OF_COUNTERS      2
#define CALLBACK_DELAY_SEC  1

int counters[NR_OF_COUNTERS] = {0,0};

void callback1() {
    counters[0]++;
}

void callback2() {
    counters[1]++;
}

int main() {
    printf("ticker callbacks test\n");
    int prev_counters[NR_OF_COUNTERS];

    ticker1.attach(&callback1, CALLBACK_DELAY_SEC);
    ticker2.attach(&callback2, CALLBACK_DELAY_SEC);

    for(;;) {
        for(int i=0; i< NR_OF_COUNTERS; i++)
        if (prev_counters[i] != counters[i]) {
            prev_counters[i] = counters[i];
            printf("%d: %d\n",i, counters[i]);
        }
    }
}

0xc0170, 2018-03-26:
@bulislaw @c1728p9 @maciejbocianski  Please review

c1728p9, 2018-04-03:
@gaborkertesz is this still a problem with the two timer implementation?

gaborkertesz, 2018-04-04:
Yes, I've just tested it with 2 timers and the issue is the same, since it's coming from the upper ticker layer, so independent of ticker HAL implementation. There is no critical issue caused by this unwanted behavior, since the system is still responsible, but overloaded with unnecessarry interval interrupts.

maciejbocianski, 2018-04-04:
@gaborkertesz How are you printing timestamp in `us_ticker_set_interrupt`, are you filtering it to be sure that collected sample is from your ticker? Or just print unconditionally?

gaborkertesz, 2018-04-04:
I'm collecting the requested timerstamps in us_ticker_set_interrupt unconditionally. Is it expected to request interrupts by other source?

maciejbocianski, 2018-04-06:
@gaborkertesz https://github.com/ARMmbed/mbed-os/pull/6515 should fix the problem with unnecessary calls of 'us_ticker_set_interrupt'

gaborkertesz, 2018-04-06:
Great! That issue really looks the same.

maciejbocianski, 2018-04-10:
@gaborkertesz please verify your issue with this fix(#6515) and close it when #6515 merged

gaborkertesz, 2018-04-10:
Okay, will do!

0xc0170, 2018-05-22:
@gaborkertesz Any update? I believe this can be closed

gaborkertesz, 2018-05-22:
As I see the referred ticket is still open: https://github.com/ARMmbed/mbed-os/pull/6515
I'm monitoring the linked issue and as soon as it's closed I'll test that fix.
Am I missing something?

cmonr, 2018-06-25:
@gaborkertesz The referenced PR has since been merged. Would you mind retesting?

0xc0170, 2018-07-03:
Please reopen with an update if there is still issue

closed 2018-07-03 by 0xc0170