ARMmbed/mbed-os #6460: ARM_CM3DS_MPS2+ drive complains No space left on device

studavekar, 2018-03-26:
### Description
- Type: Bug 
- Priority: Major 

---------------------------------------------------------------
## Bug

**Target**
ARM_CM3DS_MPS2

**Toolchain:**
GCC_ARM|ARM|IAR

**mbed-cli version:** 
1.5.0

**mbed-os sha:**
d97d55a Merge pull request #6450 from 0xc0170/fix_#6449

**Expected behavior**
During the execution of tests, the device storage should not be used up.

**Actual behavior**
we see  No space left on device error 

**Steps to reproduce**
Running tests in loop 
mbed test -m ARM_CM3DS_MPS2 -t GCC_ARM -v

Snippet of exception 

- No space left on device: '/mnt/pci-0000_00_14_0-usb-0_7_1_0-scsi-0_0_0_0/reboot.txt

```
[1522080032.80][SERI][INF] serial(port=/dev/ttyUSB0, baudrate=9600, read_timeout=0.01, write_timeout=5)
[1522080032.80][SERI][INF] reset device using 'reboot.txt' plugin...
Process Process-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/home/arm/repo/htrun/mbed_host_tests/host_tests_conn_proxy/conn_proxy.py", line 145, in conn_process
    connector = conn_primitive_factory(conn_resource, config, event_queue, logger)
  File "/home/arm/repo/htrun/mbed_host_tests/host_tests_conn_proxy/conn_proxy.py", line 105, in conn_primitive_factory
    config=config)
  File "/home/arm/repo/htrun/mbed_host_tests/host_tests_conn_proxy/conn_primitive_serial.py", line 75, in __init__
    self.reset_dev_via_serial(delay=self.forced_reset_timeout)
  File "/home/arm/repo/htrun/mbed_host_tests/host_tests_conn_proxy/conn_primitive_serial.py", line 92, in reset_dev_via_serial
    polling_timeout=self.config.get('polling_timeout'))
  File "/home/arm/repo/htrun/mbed_host_tests/host_tests_plugins/__init__.py", line 86, in call_plugin
    return HOST_TEST_PLUGIN_REGISTRY.call_plugin(type, capability, *args, **kwargs)
  File "/home/arm/repo/htrun/mbed_host_tests/host_tests_plugins/host_test_registry.py", line 69, in call_plugin
    return plugin.execute(capability, *args, **kwargs)
  File "/home/arm/repo/htrun/mbed_host_tests/host_tests_plugins/module_reset_mps2.py", line 83, in execute
    reboot_fh = open(reboot_file_path, "w")
IOError: [Errno 28] No space left on device: '/mnt/pci-0000_00_14_0-usb-0_7_1_0-scsi-0_0_0_0/reboot.txt'
```

- free space on device /mnt/pci-0000_00_14_0-usb-0_7_1_0-scsi-0_0_0_0
```
Filesystem      Size  Used Avail Use% Mounted on
udev            7.8G     0  7.8G   0% /dev
tmpfs           1.6G  154M  1.5G  10% /run
/dev/sda4       446G  166G  257G  40% /
tmpfs           7.8G  172K  7.8G   1% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           7.8G     0  7.8G   0% /sys/fs/cgroup
tmpfs           1.6G   72K  1.6G   1% /run/user/1000
/dev/nvme0n1p1  256M   34M  223M  13% /boot/efi
/dev/nvme0n1p4 1000M  371M  630M  38% /mnt/A40E67670E67318A
/dev/sda2       463G  191M  463G   1% /mnt/pci-0000_00_17_0-ata-1
/dev/sdd        2.1M  8.0K  2.1M   1% /mnt/pci-0000_00_14_0-usb-0_6_2_1_1_1-scsi-0_0_0_0
/dev/sdb1       1.9G  8.3M  1.9G   1% /mnt/pci-0000_00_14_0-usb-0_7_1_0-scsi-0_0_0_0
```
- FIle on device 
```
arm@austin-ci-linux-002:~/repo/mbed-os$ du -sh /mnt/pci-0000_00_14_0-usb-0_7_1_0-scsi-0_0_0_0/*
32K     /mnt/pci-0000_00_14_0-usb-0_7_1_0-scsi-0_0_0_0/config.txt
6.0M    /mnt/pci-0000_00_14_0-usb-0_7_1_0-scsi-0_0_0_0/MB
288K    /mnt/pci-0000_00_14_0-usb-0_7_1_0-scsi-0_0_0_0/mbed.elf
32K     /mnt/pci-0000_00_14_0-usb-0_7_1_0-scsi-0_0_0_0/mbed.htm
1.9M    /mnt/pci-0000_00_14_0-usb-0_7_1_0-scsi-0_0_0_0/SOFTWARE
96K     /mnt/pci-0000_00_14_0-usb-0_7_1_0-scsi-0_0_0_0/System Volume Information

arm@austin-ci-linux-002:/mnt$ df -i /mnt/pci-0000_00_14_0-usb-0_7_1_0-scsi-0_0_0_0
Filesystem     Inodes IUsed IFree IUse% Mounted on
/dev/sdb1           0     0     0     - /mnt/pci-0000_00_14_0-usb-0_7_1_0-scsi-0_0_0_0

```

- this works by deleting a file on the drive 
```
arm@austin-ci-linux-002:/mnt/pci-0000_00_14_0-usb-0_7_1_0-scsi-0_0_0_0$ touch reboot.txt
touch: cannot touch 'reboot.txt': No space left on device

arm@austin-ci-linux-002:/mnt/pci-0000_00_14_0-usb-0_7_1_0-scsi-0_0_0_0$ ls
config.txt  MB  mbed.elf  mbed.htm  SOFTWARE  System Volume Information

arm@austin-ci-linux-002:/mnt/pci-0000_00_14_0-usb-0_7_1_0-scsi-0_0_0_0$ rm mbed.elf

arm@austin-ci-linux-002:/mnt/pci-0000_00_14_0-usb-0_7_1_0-scsi-0_0_0_0$ touch reboot.txt
```

cc : @ashok-rao @hug-dev

ashok-rao, 2018-03-26:
@tkaman / @hug-dev ..could you please take a look at this?

hug-dev, 2018-03-27:
Hi @studavekar 
I will look into this and try to see what we can do!

tkaman, 2018-04-04:
Hi @studavekar,
I could not reproduce the problem so far.
How often did you see this? Only once or many times?
What I experienced is that the D drive disconnected, and did not reconnected until I manually reseted the board. Did you experienced this issue, or just the no space left problem?

Which HT version are you using? I am using 1.3.1.
As I see you are using Linux. Did you try this with Windows as well?

Anyway, this does not seem like a HAL issue, it is rather a MPS2 board or maybe an environment problem.

hug-dev, 2018-04-05:
I could not reproduce this issue as well (having executed around 15 tests).
The only thing I could see sometimes is that after copying the `reboot.txt` file and after the board reboots, it is not automounted by the OS anymore, I had to manually mount it or in the worst case reset the board manually (pushing the `RESET` button).

Configuration:
* Ubuntu 16.04.4 LTS
* `mbedgt` version `1.3.3`
* `mbedls` version `1.4.1`
* `mbedhtrun` version `1.3.1`
* MPS2+ firmware version `2.2.6`

ashok-rao, 2018-04-13:
@studavekar ..bump!

studavekar, 2018-05-01:
Did an overnight test with 2 boards didn't see  `No space left on device` issue. However tests were stopped stalled because of device failed to reboot using reboot,txt.

tkaman, 2018-05-15:
@ashok-rao , @studavekar 
Can we close this ticket now? 
Looks like the problem is seen only once, so we can't make progress on this.

For the reboot issue you can open a new ticket, although it should be investigated by the MPS2 team, as it does not look like a CM3DS issue.
Let me know if you need anything from our side.