ARMmbed/mbed-os #6476: mbed-os is not compiling if BLE and Cordio features are enabled

KariHaapalehto, 2018-03-27:
### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
MTB_ADV_WISE_1530

**Toolchain:**
all toolchains GCC_ARM, ARM, IAR

**Expected behavior**
mbed-os should compile with BLE and Cordio features enabled

**Actual behavior**
11:34:46 Compile [  6.5%]: mbed_shared_queues.cpp
11:34:46 Compile [  6.6%]: CordioHCIDriver.cpp
11:34:46 Compile [  6.8%]: CordioBLE.cpp
11:34:46 [Fatal Error] MemorySecurityDB.h@20,24: SecurityDB.h: No such file or directory
11:34:47 [ERROR] In file included from ./mbed-os/features/FEATURE_BLE/targets/TARGET_CORDIO/CordioBLE.h:33:0,
11:34:47                  from ./mbed-os/features/FEATURE_BLE/targets/TARGET_CORDIO/source/CordioBLE.cpp:34:
11:34:47 ./mbed-os/features/FEATURE_BLE/ble/pal/MemorySecurityDB.h:20:24: fatal error: SecurityDB.h: No such file or directory
11:34:47  #include "SecurityDB.h"
11:34:47                         ^
11:34:47 compilation terminated.
11:34:47 
11:34:47 [mbed] ERROR: "/usr/bin/python" returned error code 1.
11:34:47 [mbed] ERROR: Command "/usr/bin/python -u /builds/ws/ipcore-cliapp-wiced-test-build/3a51a639/cliapp/mbed-os/tools/make.py -t GCC_ARM -m MTB_ADV_WISE_1530 --source . --build ./BUILD/MTB_ADV_WISE_1530/GCC_ARM --app-config mbed_app.wlan0.json" in "/builds/ws/ipcore-cliapp-wiced-test-build/3a51a639/cliapp"

----------------------------------------------------------------
## Solution
MemorySecurityDB.h try to include "SecurityD**B**.h", but in github the file is named as "SecurityD**b**.h"
GenericSecurityManager.h: include correctly "ble/pal/SecurityD**b**.h"

Please either rename "SecurityD**b**.h" as "SecurityD**B**.h" and correct the file include at GenericSecurityManager.h OR correct the file include at MemorySecurityDB.h

0xc0170, 2018-03-27:
@pan- @donatieng Please review

donatieng, 2018-03-28:
@KariHaapalehto @0xc0170 Thanks, fix on its way

pan-, 2018-04-04:
@KariHaapalehto can we close this issue ?

KariHaapalehto, 2018-04-04:
Yes, issue can be closed

closed 2018-04-04 by 0xc0170