ARMmbed/mbed-os #6478: compiling MODULE_UBLOX_ODIN_W2 fails

netanelgonen, 2018-03-27:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Related issue: `#abc`
- Priority: Blocker

while trying to compile MODULE_UBLOX_ODIN_W2 target the compiler does not found the PinNames.h file.

workaround: 
cd targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_MODULE_UBLOX_ODIN_W2
cp ./TARGET_MTB_UBLOX_ODIN_W2/PinNames.h  ./

---------------------------------------------------------------
## Bug

**Target**
MODULE_UBLOX_ODIN_W2

**Toolchain:**
GCC_ARM

0xc0170, 2018-03-27:
I believe `MODULE_UBLOX_ODIN_W2` is not valid target name. This "parent" is a module that is sharing some common properties. Please look at odin, mtb or connect odin targets - those are actually targets that inherit from module ublox odin

netanelgonen, 2018-03-28:
When trying to compile on the mbed-os#062164eaad9c36299a9b30eaac9077b8c02472a5 this is listed as one of the targets...
should it be removed? the error does not reproduce while using target name MTB_UBLOX_ODIN_W2

0xc0170, 2018-03-28:
> When trying to compile on the mbed-os#062164eaad9c36299a9b30eaac9077b8c02472a5 this is listed as one of the targets...

How is it reported? Can you share the exact steps to reproduce? 
It should not be as it is not complete target

cc @theotherjimmy

netanelgonen, 2018-03-28:
to reproduce:
1. clone mbed-os
2. checkout master & pull
3. mbed compile -t GCC_ARM -m AAA
getting an output message:
usage: make.py [-h] [-m MCU] [-t TOOLCHAIN] [--color] [--cflags CFLAGS]
               [--asmflags ASMFLAGS] [--ldflags LDFLAGS] [-c]
               [--profile PROFILE] [--app-config APP_CONFIG] [-p PROGRAM]
               [-n PROGRAM] [-j JOBS] [-v] [--silent] [-D MACROS]
               [-S [{matrix,toolchains,targets}]] [-f GENERAL_FILTER_REGEX]
               [--stats-depth STATS_DEPTH] [--automated] [--host HOST_TEST]
               [--extra EXTRA] [--peripherals PERIPHERALS]
               [--dep DEPENDENCIES] [--source SOURCE_DIR]
               [--duration DURATION] [--build BUILD_DIR] [-N ARTIFACT_NAME]
               [-d DISK] [-s SERIAL] [-b BAUD] [-L] [--rpc] [--usb] [--dsp]
               [--testlib] [--build-data BUILD_DATA] [-l LINKER_SCRIPT]
make.py: error: argument -m/--mcu: AAA is not a supported MCU. Supported MCUs are:
ARCH_BLE,              ARCH_BLE_BOOT,         ARCH_BLE_OTA, 
ARCH_GPRS,             ARCH_LINK,             ARCH_LINK_BOOT, 
ARCH_LINK_OTA,         ARCH_MAX,              ARCH_PRO, 
ARM_BEETLE_SOC,        ARM_CM3DS_MPS2,        ARM_IOTSS_BEID, 
ARM_MPS2_M0,           ARM_MPS2_M0P,          ARM_MPS2_M1, 
ARM_MPS2_M3,           ARM_MPS2_M4,           ARM_MPS2_M7, 
B96B_F446VE,           BLUEPILL_F103C8,       DELTA_DFBM_NQ620, 
DELTA_DFCM_NNN40,      DELTA_DFCM_NNN40_BOOT, DELTA_DFCM_NNN40_OTA, 
DELTA_DFCM_NNN50,      DELTA_DFCM_NNN50_BOOT, DELTA_DFCM_NNN50_OTA, 
DISCO_F051R8,          DISCO_F100RB,          DISCO_F303VC, 
DISCO_F334C8,          DISCO_F401VC,          DISCO_F407VG, 
DISCO_F413ZH,          DISCO_F429ZI,          DISCO_F469NI, 
DISCO_F746NG,          DISCO_F769NI,          DISCO_L053C8, 
DISCO_L072CZ_LRWAN1,   DISCO_L475VG_IOT01A,   DISCO_L476VG, 
DISCO_L496AG,          EFM32GG_STK3700,       EFM32HG_STK3400, 
EFM32LG_STK3600,       EFM32PG12_STK3402,     EFM32PG_STK3401, 
EFM32WG_STK3800,       EFM32ZG_STK3200,       ELEKTOR_COCORICO, 
ELMO_F411RE,           EV_COG_AD3029LZ,       EV_COG_AD4050LZ, 
FF1705_L151CC,         FF_LPC546XX,           GR_LYCHEE, 
HEXIWEAR,              HRM1017,               HRM1017_BOOT, 
HRM1017_OTA,           K20D50M,               K22F, 
K64F,                  K66F,                  K82F, 
KL05Z,                 KL25Z,                 KL26Z, 
KL27Z,                 KL43Z,                 KL46Z, 
KL82Z,                 KW24D,                 KW41Z, 
LPC1114,               LPC11C24,              LPC11U24, 
LPC11U24_301,          LPC11U34_421,          LPC11U35_401, 
LPC11U35_501,          LPC11U35_501_IBDAP,    LPC11U35_Y5_MBUG, 
LPC11U37H_401,         LPC11U37_501,          LPC11U68, 
LPC1347,               LPC1549,               LPC1768, 
LPC1769,               LPC4088,               LPC4088_DM, 
LPC4330_M0,            LPC4330_M4,            LPC4337, 
LPC54114,              LPC546XX,              LPC810, 
LPC812,                LPC824,                LPCCAPPUCCINO, 
MAX32600MBED,          MAX32620HSP,           MAX32625MBED, 
MAX32625NEXPAQ,        MAX32630FTHR,          MAXWSNENV, 
MBED_CONNECT_ODIN,     MCU_LPC4088,           MCU_LPC546XX, 
MICRONFCBOARD,         MIMXRT1050_EVK,       ** MODULE_UBLOX_ODIN_W2**, 
MOTE_L152RC,           MTB_ADV_WISE_1510,     MTB_ADV_WISE_1570, 
MTB_LAIRD_BL600,       MTB_LAIRD_BL652,       MTB_MTS_DRAGONFLY, 
MTB_MTS_XDOT,          MTB_MURATA_ABZ,        MTB_MXCHIP_EMW3166, 
MTB_UBLOX_NINA_B1,     MTB_UBLOX_ODIN_W2,     MTM_MTCONNECT04S, 
MTM_MTCONNECT04S_BOOT, MTM_MTCONNECT04S_OTA,  MTS_DRAGONFLY_F411RE, 
MTS_GAMBIT,            MTS_MDOT_F405RG,       MTS_MDOT_F411RE, 
NCS36510,              NRF51822,              NRF51822_BOOT, 
NRF51822_OTA,          NRF51822_Y5_MBUG,      NRF51_DK, 
NRF51_DK_BOOT,         NRF51_DK_LEGACY,       NRF51_DK_OTA, 
NRF51_DONGLE,          NRF51_DONGLE_BOOT,     NRF51_DONGLE_LEGACY, 
NRF51_DONGLE_OTA,      NRF51_MICROBIT,        NRF51_MICROBIT_B, 
NRF51_MICROBIT_BOOT,   NRF51_MICROBIT_B_BOOT, NRF51_MICROBIT_B_OTA, 
NRF51_MICROBIT_OTA,    NRF52840_DK,           NRF52_DK, 
NUCLEO_F030R8,         NUCLEO_F031K6,         NUCLEO_F042K6, 
NUCLEO_F070RB,         NUCLEO_F072RB,         NUCLEO_F091RC, 
[... 9 lines cut ...]
NUCLEO_L486RG,         NUCLEO_L496ZG,         NUCLEO_L496ZG_P, 
NUMAKER_PFM_M453,      NUMAKER_PFM_M487,      NUMAKER_PFM_NANO130, 
NUMAKER_PFM_NUC472,    NZ32_SC151,            OC_MBUINO, 
OSHCHIP,               RBLAB_BLENANO,         RBLAB_BLENANO2, 
RBLAB_BLENANO_BOOT,    RBLAB_BLENANO_OTA,     RBLAB_NRF51822, 
RBLAB_NRF51822_BOOT,   RBLAB_NRF51822_OTA,    REALTEK_RTL8195AM, 
RO359B,                RZ_A1H,                RZ_A1XX, 
SAKURAIO_EVB_01,       SAMD21G18A,            SAMD21J18A, 
SAMG55J19,             SAML21J18A,            SAMR21G18A, 
SARA_NBIOT_EVK,        SEEED_TINY_BLE,        SEEED_TINY_BLE_BOOT, 
SEEED_TINY_BLE_OTA,    SILICA_SENSOR_NODE,    SSCI824, 
STEVAL_3DP001V1,       TB_SENSE_1,            TB_SENSE_12, 
TEENSY3_1,             TMPM066,               TY51822R3, 
TY51822R3_BOOT,        TY51822R3_OTA,         UBLOX_C027, 
UBLOX_C030_N211,       UBLOX_C030_R410M,      UBLOX_C030_U201, 
UBLOX_EVA_NINA,        UBLOX_EVK_NINA_B1,     UBLOX_EVK_ODIN_W2, 
UBRIDGE,               USENSE,                VBLUNO51, 
VBLUNO51_BOOT,         VBLUNO51_LEGACY,       VBLUNO51_OTA, 
VBLUNO52,              VK_RZ_A1H,             WALLBOT_BLE, 
WALLBOT_BLE_BOOT,      WALLBOT_BLE_OTA,       WIZWIKI_W7500, 
WIZWIKI_W7500ECO,      WIZWIKI_W7500P,        XADOW_M0, 
XBED_LPC1768,          XDOT_L151CC            
[mbed] ERROR: "/usr/bin/python" returned error code 2.
[mbed] ERROR: Command "/usr/bin/python -u /home/netgon01/proj/enable/mbed-os/tools/make.py -t GCC_ARM -m AAA --source . --build ./BUILD/AAA/GCC_ARM" in "/home/netgon01/proj/enable/mbed-os"
---

netanelgonen, 2018-03-28:
then compiling with the  MODULE_UBLOX_ODIN_W2 target

netanelgonen, 2018-03-28:
change the name of the issue as the 1st issue not exists

theotherjimmy, 2018-03-28:
@0xc0170 MODULE_... needs the key "pubilc" set to `false` in targets.json.

netanelgonen, 2018-03-29:
I have created this PR to solve this issue
https://github.com/ARMmbed/mbed-os/pull/6505

netanelgonen, 2018-04-11:
as #6505 was merged I closing this issue

closed 2018-04-11 by netanelgonen