ARMmbed/mbed-os #6495: Without TRNG and Ethernet : mbed-os-example-client using LPC1347

prvthkr, 2018-03-28:
### Description
- Type: Question
- Priority: Medium
---------------------------------------------------------------
## Question
**Target**
LPC1347

**Toolchain**
ARM

Requirement is to port [mbed-os-example-client](https://github.com/ARMmbed/mbed-os-example-client) on target LPC1347.
1. However, TRNG is not supported by this target.
    - is it possible to use [mbed-os-example-client](https://github.com/ARMmbed/mbed-os-example-client) without TRNG?
2. Also, ethernet port is not available in this target. External [Ethernet shield](https://os.mbed.com/components/W5500-Ethernet-Kit-for-IoT/)(SPI interface) to be used.
    - how to use this external ethernet shield for this [mbed-os-example-client](https://github.com/ARMmbed/mbed-os-example-client)

0xc0170, 2018-03-28:
@teetak01 Can you review?

prvthkr, 2018-03-29:
@teetak01 I'm stuck on this issue so, would you like to help me resolve this issue.

teetak01, 2018-04-05:
@JanneKiiskila

prvthkr, 2018-04-05:
Hi @teetak01 , 

>The specifications of the board seems to be way too low for Mbed OS. Can you even run mbed-os-example-blinky on that?

Ok will check.

Can you confirm me if I'll use [TARGET_TMPM46B](https://os.mbed.com/platforms/TMPM46B/), then [mbed-os-example-client](https://github.com/ARMmbed/mbed-os-example-client) using external [Ethernet Shield](https://os.mbed.com/components/W5500-Ethernet-Kit-for-IoT/) is possible or not?

JanneKiiskila, 2018-04-05:
I'm not aware of any trials we would have done with Ethernet shields, but you should try it out and if there is a working Mbed OS driver -> should work. But, we have not verified any of those ourselves.

You might want to give the networking a run for the money 1st with something like [mbed-stress-test](https://github.com/ARMmbed/mbed-stress-test) to see that it works reliably, but in general I would be tempted to recommend some other board that already has Ethernet natively in it.

JanneKiiskila, 2018-04-05:
Hi, 

the board TMPM46B seems to have plenty of RAM and flash (512 kB, 1 MB) - but it does not have TRNG. TRNG is a requirement for production level devices (for now), but we are working on a solution for [Mbed Cloud Client](https://github.com/ARMmbed/mbed-cloud-client-example).

prvthkr, 2018-04-06:
Hi @JanneKiiskila ,

> I'm not aware of any trials we would have done with Ethernet shields, but you could try it out and if there is a working Mbed OS driver -> should work.

Thanks for your information. I'll try.

>Actually, are you porting the whole board to Mbed OS? 

No, I'm not porting whole board to Mbed OS. TMPM46B already present in [mbed-os-master](https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_TOSHIBA).

>I can't locate that board from the list of [Mbed OS supported boards](https://os.mbed.com/platforms/).

Yes, this board is not in the list but drivers are present in the [mbed-os-master](https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_TOSHIBA).

>the board TMPM46B seems to have plenty of RAM and flash (512 kB, 1 MB) - but it does not have TRNG. TRNG is a requirement for production level devices..

TRNG is present in the [TMPM46B](https://os.mbed.com/platforms/TMPM46B/#tmpm46b-features).

closed 2018-04-10 by prvthkr