ARMmbed/mbed-os #6506: Online compiler error while exporting to external IDE

prvthkr, 2018-03-29:
### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
K64F|LPC1768

**Toolchain:**
GCC_ARM|ARM|IAR...Online Compiler

**Expected behavior**
Should be export.

**Actual behavior**
Giving error while exporting.

[image]

**Steps to reproduce**
Go to the online compiler and try to export any program.

theotherjimmy, 2018-04-05:
@prvthkr You're going to have to give me some more details, like what your exporting and what your target IDE is, before I can progress this issue.

prvthkr, 2018-04-06:
>what your exporting and what your target IDE is, before I can progress this issue.

Hi @theotherjimmy , Any program I'm not able to export. Target IDE is uvision.

theotherjimmy, 2018-04-06:
Thanks. That's a known bug in the online exporters that should be cleared up in the next week or so. related to #6428

prvthkr, 2018-04-07:
Thank u

closed 2018-04-07 by prvthkr