ARMmbed/mbed-os #6513: NUCLEO_L476RG: Hang up after wakeup from deepsleep.

maskedw, 2018-03-30:
### Description
- Type: Bug 
- Related issue: #5830

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_L476RG

**Toolchain:**
GCC_ARM

**Toolchain version:**
arm-none-eabi-gcc 6.3.1

**mbed-cli version:**
1.5.0

**mbed-os sha:**
addec7b

**Steps to reproduce**

The cycle to hang is irregular. It occurs approximately 1000 times or less repeatedly.
Is there a prospect of resolving ST_INTERNAL_REF 42806?

```c++
#include "mbed.h"

DigitalOut myled(LED1);
Serial pc(USBTX, USBRX);
LowPowerTimeout timeout;
static void NullFunc() {}

int main()
{
    pc.baud(115200);
    int i = 0;
    while(1)
    {
        wait_ms(10);
        printf("i = %d\n", i++);
        wait_ms(10);
        timeout.attach_us(callback(NullFunc), 10 * 1000);
        hal_deepsleep();
    }

    return 0;
}
```

jeromecoutant, 2018-03-30:
Hi

> Is there a prospect of resolving ST_INTERNAL_REF 42806?

Yes, we have worked on this issue this week.
Tests are on going.

@LMESTM

maskedw, 2018-03-30:
Great. I will wait for this problem to be resolved and close the issue.

LMESTM, 2018-04-03:
@maskedw 
would you please try out the commit here and let me know if this helps ?
LMESTM/mbed@bbc01b8

maskedw, 2018-04-04:
I ran about 1 hour. So far so good. I will continue to try it for a 2,3 days.

maskedw, 2018-04-04:
When repeating the loop, the cycle became obviously long.
1 to 2 seconds cycle?
It is assumed that PLL initialization fails and it is operating at 1 MHz set by ForceClockOutofDeepSleep().

**UART output after abnormality occurs**

```
i = 184272
i = 184273
ｹ・
￠ｺ｢Rｹ・
￠ｺｪRｱ・
￠ｺｲRｹ・
￠ｺｺRｹ・
￠ｺﾂRｹ・
￠ｺﾊRｹ・
...
```

LMESTM, 2018-04-04:
@maskedw Ok I will let it run and see if I reproduce.

LMESTM, 2018-04-05:
I have traced down the timestamp that is programmed by LP Ticker driver to the hal API 
void lp_ticker_set_interrupt(timestamp_t timestamp)

I'm tracing it in the printf instead of the incremented i variable. 
```
ts = 0x6FFF191F
ts = 0x6FFF736E
ts = 0x6FFFCDBD
ts = 0x    20E2
ts = 0x  275A4B
ts = 0x  4EB4D3
ts = 0x  645A4B
ts = 0x  8BB4D3
ts = 0x  A15A4B
ts = 0x  C8B4D3
ts = 0x  DE5A4B
ts = 0x 105B4D3
```
Note that when booting the initial default timestamp is  ts = 0x70000000
so the issue happens when we've done a complete wrap-around.

But then we need to understand why this wraps-up to 0 at this stage and not continue to 0x70000000 ...

LMESTM, 2018-04-05:
Going deeper in debug, here is what I observe. 
Standard behavior of the system in this test is:
1) lp ticker driver read ticker timestamp
2) lp ticker driver sets now the 10ms timeout timestamp 
3) enter deepsleep
4) exit deep sleep, calls null callback, set default wake-up in the future as no more ticker is programmed (current timestamp + 0X7000000) 
5) printf
6) back to 1)

Here are the traces I've taken 
```
prg ts at wake-up= 0x6FFEF5DB
ts prg bf sleep = 0xFFFEEEB1
last read after wake-up= 0xFFFF633A
prg ts at wake-up= 0x6FFF633A
ts prg bf sleep= 0xFFFF5C10
last read after wake-up= 0xFFFFD099
prg ts at wake-up= 0x6FFFD099
ts prg bf sleep= 0xFFFFC96F
last read after wake-up= 0x    1075
prg ts at wake-up= 0x    36CE
ts prg bf sleep= 0x6FFFD099
last read after wake-up= 0x  274702
prg ts at wake-up= 0x  276D5B
ts prg bf sleep= 0x70274594
last read after wake-up= 0x  4EB49A
prg ts at wake-up= 0x  4EDAF3
ts prg bf sleep= 0x704EB32C
last read after wake-up= 0x  644702
prg ts at wake-up= 0x  646D5B
ts prg bf sleep= 0x70644594
```

@0xc0170 there seems to be something in the list of events in the tickers queue. The order has changed after wrap-around, as if the first one is now the MAX event  ... is this something that have been seen already ?

@maskedw This piece of code in mbed-os and in STM32 HAL layer is being modified in feature branch here 
https://github.com/ARMmbed/mbed-os/tree/feature-hal-spec-ticker
So anything we do on master now would be replaced by this new feature branches very soon. I propose to test the same on master branch and see if the problem is still there.

LMESTM, 2018-04-06:
@maskedw 
I pushed a delay after deepsleep for now in #6561 because it's better to let everyone benefit of this fix for now. I think the issue fo slow down after 1 hour is a separate issue as the one in the title (system hangs) - what would do you think about closing that one and opening a new one for the 1 hour issue ?

maskedw, 2018-04-09:
Please try to operate at MSI 48 MHz. In this case, why does not the loop cycle become long?

```c++
uint8_t SetSysClock_MSI(void)
{
    RCC_ClkInitTypeDef RCC_ClkInitStruct;
    RCC_OscInitTypeDef RCC_OscInitStruct;

    memset(&RCC_ClkInitStruct, 0, sizeof(RCC_ClkInitStruct));
    memset(&RCC_OscInitStruct, 0, sizeof(RCC_OscInitStruct));

    RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_LSE;
    RCC_OscInitStruct.PLL.PLLState   = RCC_PLL_NONE; // No PLL update
    RCC_OscInitStruct.LSEState       = RCC_LSE_ON; // External 32.768 kHz clock on OSC_IN/OSC_OUT
    if (HAL_RCC_OscConfig(&RCC_OscInitStruct) == HAL_OK) {
        RCC->CR |= RCC_CR_MSIPLLEN; // Enable MSI PLL-mode
    }

    // MSI ON, PLL OFF
    RCC_OscInitStruct.OscillatorType       = RCC_OSCILLATORTYPE_MSI | RCC_OSCILLATORTYPE_HSI;
    RCC_OscInitStruct.HSIState             = RCC_HSI_OFF;
    RCC_OscInitStruct.MSIState             = RCC_MSI_ON;
    RCC_OscInitStruct.MSIClockRange        = RCC_MSIRANGE_11; // MSI 48MHz
    RCC_OscInitStruct.MSICalibrationValue  = RCC_MSICALIBRATION_DEFAULT;
    RCC_OscInitStruct.PLL.PLLState         = RCC_PLL_OFF;
    if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK) {
        return 0; // FAIL
    }

    RCC_ClkInitStruct.ClockType      = (RCC_CLOCKTYPE_SYSCLK | RCC_CLOCKTYPE_HCLK | RCC_CLOCKTYPE_PCLK1 | RCC_CLOCKTYPE_PCLK2);
    RCC_ClkInitStruct.SYSCLKSource   = RCC_SYSCLKSOURCE_MSI;
    RCC_ClkInitStruct.AHBCLKDivider  = RCC_SYSCLK_DIV1;
    RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV1;
    RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV1;
    if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_2) != HAL_OK) {
        return 0; // FAIL
    }

    RCC_OscInitStruct.OscillatorType       = RCC_OSCILLATORTYPE_HSI;
    RCC_OscInitStruct.HSIState             = RCC_HSI_OFF;
    if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK) {
        return 0; // FAIL
    }

    return 1;
}
```

LMESTM, 2018-04-09:
@maskedw I agree that there is still an issue, but the system does not hang anymore - so this is a different issue. I don't know the rootcause for the issue, but what I have obsserved is that the events in the LP ticker queue are inverted after some time. This might be related to a timing issue that doesn't happen when MSI is 48MHz. It might be because wake-up without PLL will be much faster, so the lost time elapsed during deep sleep will be shorter than wehn using PLL. This is just a guess for now, I will need to debug and test more ...

maskedw, 2018-04-10:
understood. It certainly should be discussed as another problem. Thank you very much.

LMESTM, 2018-04-10:
@maskedw 
Thank you. Have you created anoter Issue to follow up on the "long loopcycle after 1 hour" ? 
Note that I have used your MSI SetSysClk function and then added a 2ms at the end, and I can reproduce the same issue of "long loop cycle"

LMESTM, 2018-04-11:
@maskedw and others that may be interested. There are ongoing changes for tickless support in mbed, in feature branches like https://github.com/ARMmbed/mbed-os/tree/feature-hal-spec-ticker, and other fixes on master. When taking those various updates, I've run the test for more than 12 hours and the "long loop cycle" issue was not reproduced. That means this will be fixed in mbed 5.9 - let me know if this sounds ok for you.