ARMmbed/mbed-os #6514: The printf immediately after wakeup from deepsleep gets garbled.

maskedw, 2018-03-30:
### Description
- Type: Bug 
---------------------------------------------------------------
## Bug

**Target**
NUCLEO_L476RG

**Toolchain:**
GCC_ARM

**Toolchain version:**
arm-none-eabi-gcc 6.3.1

**mbed-cli version:**
1.5.0

**mbed-os sha:**
addec7b

**Steps to reproduce**

```c++
#include "mbed.h"

DigitalOut myled(LED1);
Serial pc(USBTX, USBRX);
LowPowerTimeout timeout;
static void NullFunc() {}

int main()
{
    pc.baud(115200);
    int i = 0;
    while(1)
    {
        // If you put a little weight, it will not be garbled
        // wait_ms(10);
        printf("i = %d\n", i++);

        // Wait for the transmission to be completed before sleeping
        wait_ms(10);
        timeout.attach_us(callback(NullFunc), 10 * 1000);
        hal_deepsleep();
    }

    return 0;
}
```

**Outputs**
```
i = 0
・= 1
・= 2
・= 3
・= 4
・= 5
・= 6
i = 7
i = 8
・= 9
・= 10
・= 11
・= 12
・= 13
・= 14
・= 15
・= 16
・= 17
・= 18
・= 19
・= 20
・= 21
...
```

jeromecoutant, 2018-03-30:
Looks same question as #6451 

cc @bulislaw @c1728p9

marcemmers, 2018-03-30:
@jeromecoutant It's not exactly the same as #6451 because the issue I am having is at the end of transmission before entering sleep. This seems to be a issue where the clocks are not yet initialized correctly.

maskedw, 2018-03-30:
This problem is right after deepsleep, so I think it is a different problem from # 6451.

LMESTM, 2018-03-30:
@maskedw we will test after we've fully validated sleep mode exit.

But your use case should work fine already ... maybe it musses a small modification. 
Could you try out to move     core_util_critical_section_exit();  call from line 91 to last line of the function ?
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/sleep.c#L91

Note that in the future, hal_deepsleep() will be called by mbed-os idle loop rather than directly from application ...

maskedw, 2018-04-02:
I made changes but it did not solve it.
```c++
    // Restart HAL tick
    HAL_ResumeTick();

    // Enable IRQs
    // core_util_critical_section_exit();

    // After wake-up from STOP reconfigure the PLL
    SetSysClock();

    TIM_HandleTypeDef TimMasterHandle;
    TimMasterHandle.Instance = TIM_MST;
    __HAL_TIM_SET_COUNTER(&TimMasterHandle, EnterTimeUS);

#if DEVICE_RTC
    /* Wait for RTC RSF bit synchro if RTC is configured */
#if (TARGET_STM32F2) || (TARGET_STM32F4) || (TARGET_STM32F7)
    if (READ_BIT(RCC->BDCR, RCC_BDCR_RTCSEL)) {
#else /* (TARGET_STM32F2) || (TARGET_STM32F4) || (TARGET_STM32F7) */
    if (__HAL_RCC_GET_RTC_SOURCE()) {
#endif  /* (TARGET_STM32F2) || (TARGET_STM32F4) || (TARGET_STM32F7) */
        rtc_synchronize();
    }
#endif
    // moved to the last line
    core_util_critical_section_exit();
}
```

LMESTM, 2018-04-03:
@maskedw  
I am working on somehow related issues and wanted to verify if this helps here. 
I tested and I confirm that my fix seems to solve your issue. I pushed my patch here:
https://github.com/LMESTM/mbed/commit/bbc01b83dd33ea413ffd0c76473e8bf8702ebe65
Please confirm if ok for you. 
Br
Laurent

maskedw, 2018-04-04:
I tried [LMESTM/mbed@bbc01b8. ](https://github.com/LMESTM/mbed/commit/bbc01b83dd33ea413ffd0c76473e8bf8702ebe65)
But, garbled characters were not improved.

LMESTM, 2018-04-04:
@maskedw that's really strange because I actually tested this. I just tried it again on master. Without the patch, I can first reproduce the issue, then I cherry-pick the commit I pointed to and the problem is solved.  Every time I reset, the program restarts fine and displays i correctly ... 
```
i = 60
i = 61
i = 62
i = 63
i = 0
i = 1
i = 2
i = 3
i = 4
i = 5
i = 6
i = 7
i = 0
i = 1
i = 2
i = 3
i = 4
i = 5
i = 6
i = 0
i = 1
i = 2
i = 3
i = 4
i = 5
i = 6
i = 7
i = 8
i = 9
i = 10
i = 11
```

maskedw, 2018-04-04:
I tried it at some baud rates.

  | baud
-- | --
O | 9600
O | 19200
X | 38400
X | 57600
X | 115200

LMESTM, 2018-04-04:
I'm working with 115200 baudrate as well. I finally got it, but it happens much more rarely than yours. 
Anyway I think it's related to MSI calibration, there are kind of hick-ups on MSI clock around 2ms after switch off.  When hick-ups happen when UART transfer is ongoing, it will corrupt the data. This seems to be due to the ongoing calibration of the MSI clock. 

This can easily be worked-around by adding some waiting time of few hundreds of micro-seconds at the end of SetSysClock (or after the call to hal_deepsleep) function, but I prefer to fully understand the case first.

LMESTM, 2018-04-06:
@maskedw  
until I get an more internal inputs about this issue, please consider adding a small delay after exiting sleep mode, like:
```
 hal_deepsleep();
wait_ms(1);
```
Note that restorging clocks and system state when exiting deepsleep takes already between 1 and 2ms.

maskedw, 2018-04-06:
I got it. I will leave this issue open.

LMESTM, 2018-04-06:
@maskedw
I pushed a delay after deepsleep for now in #6561 - this should fix the garbled character issue - let me know ...

maskedw, 2018-04-09:
Thanks, issue has been fixed in #6561

  | baud
-- | --
GOOD | 9600
GOOD | 19200
GOOD | 38400
GOOD | 57600
GOOD | 115200
GOOD | 230400
GOOD | 460800
GOOD | 921600

closed 2018-04-09 by maskedw