ARMmbed/mbed-os #6521: mbed_error_vfprintf retargeting

pavels, 2018-04-01:
## Enhancement

We now have retargeting for stdout and stderr which is greate. We can use SWO for this stuff pretty easy and that is realy greate.

There is one thing left behind
`mbed_error_vfprintf` (in `/mbed-os/platform/mbed_board.c`)

This is used to print out debug messages when the firmware crashes. It would be nice if this thing can somehow use the retargeting system as well (so we can send error messages to SWO as well)

I am thinking something like this 
```
void mbed_error_vfprintf(const char * format, va_list arg) {
    #define ERROR_BUF_SIZE      (128)
    core_util_critical_section_enter();
    char buffer[ERROR_BUF_SIZE];
    int size = vsnprintf(buffer, ERROR_BUF_SIZE, format, arg);
    if (size > 0) {
        for (int i = 0; i < size; i++) {
            putc(buffer[i], stderr);
        }
    }
    core_util_critical_section_exit();
}
```

I can make PR if it is OK, but maybe it is not, maybe there is some side effects i am missing. What do you think?

0xc0170, 2018-04-03:
> We now have retargeting for stdout and stderr which is greate. We can use SWO for this stuff pretty easy and that is realy greate.

Thanks for the feedback

> I can make PR if it is OK, but maybe it is not, maybe there is some side effects i am missing. What do you think?

@kjbracey-arm Please review

kjbracey-arm, 2018-04-03:
Hi, yes, we've discussed this - it's not quite that simple, alas. (Which is my excuse for not having done it).

`putc` calls into the C library, which in at least some cases has an internal mutex for each file handle, including stdout/stderr.

As such, you can't use those calls from the sort of tricky contexts this is issued from - critical section as you can see, and possibly in an interrupt or exception handler. That's why it doesn't already call `putc`, and has the direct `serial_xxx` HAL calls.

But what you should now be able to do is `write(STDERR_FILENO, buffer, size)`. That will be fine, _as long as the `FileHandle` providing `stderr` is safe to use in a critical section_.  Effectively that would be a requirement for the retargeting - if you want to see these mbed_errors, your `FileHandle::write` needs to work from critical section inside exceptions.

As it happens, the default `DirectSerial` used when `platform.stdio-buffered-serial=false` is okay, but the buffered `UARTSerial` used when `platform.stdio-buffered-serial=true` isn't. (I don't know about the SWO).

`DirectSerial` gets away with it because base serial is so dumb - spinning while not readable works just as well (badly) in thread and exception context.

For `UARTSerial`, it's a combination of the API mutex, and the actual thread sleep mechanism used to achieve blocking.

We can lose the pointless API mutex easily enough for `UARTSerial` - already done in one draft commit here: https://github.com/kjbracey-arm/mbed-os/commits/cv_cs

But we'd need to add special case code for `if (blocking && in critical section)` to make that write work, making it fall back to "dumb" behaviour - manually drain anything remaining in the write buffer, then write out this data synchronously.

Feel free to put together a PR along these lines - 3 commits, taking that 1 from me to remove the API mutex, then some logic for `UARTSerial::write`, then a patch to the `mbed_error_vfprintf`.

pavels, 2018-04-05:
Hi
@kjbracey-arm  i will try to do that - i have one question "in critical section" - that means checking if in critical section or in interrupt right? because this can be also called from interrupt context and should work

kjbracey-arm, 2018-04-09:
Yes, basically any non-normal-thread context. I think those cases enumerate to:

* in exception (eg data abort)
* in interrupt (including supervisor call)
* in thread context but with interrupts disabled (aka critical section)

Never been clear what the most bullet-proof portable (including ARM-A and ARM-M) mechanism to figure that out is in mbed OS 2/5. For M it's PRIMASK and interrupt priority; for A it's CPSR:I and mode. There are various calls in various headers - have a look.

SenRamakri, 2018-07-23:
@pavels , @kjbracey-arm - Thanks for looking into this and the ideas. According to the discussions above, we know what needs to be done and I'm thinking a new PR with the above said enhancements needs to be created. Since there are no specific issues to be fixed here I'm closing this issue for now.

closed 2018-07-23 by SenRamakri