ARMmbed/mbed-os #6526: failure to build with `target.stack_size`

drahnr, 2018-04-03:
## Bug

**Target**
NRF52

**Toolchain:**
GCC_ARM

**mbed-cli version:** 
(`mbed --version`)

**mbed-os sha:**
5.8.1

**Expected behavior**
compile like mbed os 5.7.1

**Actual behavior**

> > mbed compile --app-config ./config/NRF52_DK/mbed_app.json --profile ./config/debug.json 
> [ERROR] in ./config/NRF52_DK/mbed_app.json element target_overrides.target.heap_size: 32768 is not of type u'object'; in ./config/NRF52_DK/mbed_app.json element target_overrides.target.stack_size: 8192 is not of type u'object'
> [mbed] ERROR: "/usr/bin/python2" returned error code 1.

**Steps to reproduce**
```
{
	"target_overrides": {
		"target.stack_size": 8192,
		"target.heap_size": 32768,
	},
	"config": {
		"*": {
			"platform.stdio-flush-at-exit": false
		}
	}
}
```

0xc0170, 2018-04-03:
@theotherjimmy Can you review please?

theotherjimmy, 2018-04-03:
@drahnr Your configuration is incorrect. the "target_overrides" section must contain target names, and config must contain new configuration variables. I think what you want is this config file:
```JSON
{
    "target_overrides": {
        "*": {
            "target.stack_size": 8192,
            "target.heap_size": 32768,
            "platform.stdio-flush-at-exit": false
        }
    }
}
```
That being said, you'll find that overriding `target.stack_size` and `target.heap_size` have no affect.

drahnr, 2018-04-04:
thanks for clarifying.

closed 2018-04-04 by drahnr