ARMmbed/mbed-os #6529: How can I use WPS on ODIN-W2?

dmelo, 2018-04-03:
### Description
- Type: Question
- Priority: Major

---------------------------------------------------------------
## Question

I'm struggling to find a way for users to configure WiFi network on ODIN-W2. ODIN-W2 does not support AP mode and I couldn't find much about WPS.

On [ODIN-W2 series Technologies Application Note](https://www.u-blox.com/sites/default/files/ODIN-W2-Technologies_ApplicationNote_%28UBX-17051357%29.pdf) it says that "Wi-Fi Protected Setup (WPS) can be used to more easily connect Wi-Fi Stations to a Wi-Fi Access Point". Is it possible to use WPS on ODIN-W2 with MBed OS? If so, does anyone have sample code?

If it is not possible to work with WPS on ODIN-W2. Does anyone have a suggestion of how to let users configure their WiFi?

0xc0170, 2018-04-03:
@ARMmbed/team-ublox Please review

magnushanssonublox, 2018-04-04:
WPS is not supported by ODIN-W2 series. The referenced document was not intended to be published and has been removed. Sorry for the inconvenience.

Regarding user configuration of WiFi you have to handle interaction with end user and provide means to input SSID and password.

0xc0170, 2018-04-10:
@dmelo Does this answer your question?