ARMmbed/mbed-os #6532: RTL8195AM: compilation fails in online compiler

yennster, 2018-04-03:
## Bug

**Target**
RTL8195AM

**Toolchain:**
Online compiler

**Expected behavior**
Programs compile successfully in the online compiler and a .bin file downloads for RTL8195AM target.

**Actual behavior**
A lot of warnings and the following errors:
```
Error: Internal error. 0
Error: Compile task failed. 242
Info: Unable to download. Fix the reported errors...
```

**Steps to reproduce**
Open the RTL8195AM board in the online compiler and try compiling blinky or the wifi example for it: https://os.mbed.com/compiler/#platform:Realtek-RTL8195AM

The same error occurs for the following programs that I've tried in the online compiler:
- [mbed-os-example-wifi](https://github.com/ARMmbed/mbed-os-example-wifi)
- [mbed-os-example-blinky](https://github.com/ARMmbed/mbed-os-example-blinky)

0xc0170, 2018-04-05:
@theotherjimmy Who could help with this?

theotherjimmy, 2018-04-05:
Duplicate of https://github.com/ARMmbed/mbed-os/issues/5976

yennster, 2018-04-05:
@theotherjimmy It also can't compile blinky

theotherjimmy, 2018-04-05:
Yep. I'm aware. I tested my fixes for #5976 With blinky too.

closed 2018-04-05 by yennster