ARMmbed/mbed-os #6539: RAK811 target and battery measurment

atoy40, 2018-04-04:
Hello,
the PA_2 pin is used as an ADC on the rak811 to get battery voltage (using a 5/3 voltage divisor)
So it's needed to define it in the adc pinmap for this target:
````
const PinMap PinMap_ADC[] = {
    {PA_2, ADC_1, STM_PIN_DATA_EXT(STM_MODE_ANALOG, GPIO_NOPULL, 0, 2, 0)},
    {PB_12, ADC_1, STM_PIN_DATA_EXT(STM_MODE_ANALOG, GPIO_NOPULL, 0, 18, 0)}, // ADC_IN18
    {PB_14, ADC_1, STM_PIN_DATA_EXT(STM_MODE_ANALOG, GPIO_NOPULL, 0, 20, 0)}, // ADC_IN20
    {PB_15, ADC_1, STM_PIN_DATA_EXT(STM_MODE_ANALOG, GPIO_NOPULL, 0, 21, 0)}, // ADC_IN21
    {NC,    NC,    0}
};
```

and to remove it from the PinMap_UART_TX.

Anthony.

ithinuel, 2018-04-05:
Hi, I confirm that PA0-2 are mapped to ADC_IN0-2 but nothing on RAK811's datasheet mention any battery voltage measurement.

atoy40, 2018-04-05:
@ithinuel May be some confusion. I'm talking about the RAK811 TrackerBoard (https://fr.aliexpress.com/item/RAK811-LoRa-Conseil-Tracker-MAX-7Q-GPS-Module-et-MEMS-Capteur-sans-fil-Distance-Positionnement-Solution/32844470946.html?shortkey=Enme2qU3&addresstype=600) where battery is output is connected to PA_2
and the MTB_RAK811 target is may be only for the RAK811 "chip-only" ? May be we need to have another tarcker for the TrackerBoard (I think it is the main usage of rak811) ?

ithinuel, 2018-04-05:
That would probably make sense indeed.
I don't know yet if we can "refine" a target even more :
- stm
  - stm32l1
      - stm32l151CUB6A
         - RAK811
           - MTB_RAK811
           - TrackerBoard

PR #6544 (2018-04-10): add ADC_AN0-2 mapped on PA_0-2
### Description

This fixes #6539 by adding supported pins to the adc peripheral

### Pull request type

- [X] Fix
- [ ] Refactor  
- [ ] New target  
- [ ] Feature  
- [ ] Breaking change
closed 2018-04-10 by cmonr