ARMmbed/mbed-os #6542: Subsequent TCPSocket connect call to invalid address succeeds

kegilbert, 2018-04-04:
Back to back connect calls on a TCPSocket to an invalid IP address/port first returns an expected failure (-3004 : NSAPI_ERROR_NO_CONNECTION). A second call to the same address though returns NSAPI_ERROR_OK (0).

Take the following code:

```
EthernetInterface net;
net.connect();
printf("IP address: %s\n", net.get_ip_address());

TCPSocket socket;
socket.open(&net);

int err = socket.connect("10.0.0.0", 80);
printf("Err: %d\r\n", err);

err = socket.connect("10.0.0.0", 80);
printf("Err: %d\r\n", err);
```

The output is:

```
IP address: 10.0.0.226
Err: -3004
Err: 0
```

The same happens when changing the second address to another invalid address.

I.E. try to connect to 10.0.0.0:80, fail. Connect to 10.0.0.1:80 succeed. Reset the device but try to connect to 10.0.0.1:80, fail. Connect to 10.0.0.2:80 succeed. Any subsequent call following a failed connect seems to succeed.

0xc0170, 2018-04-05:
@kjbracey-arm can you please review?

kjbracey-arm, 2018-04-05:
Necessary information: which network device/stack?

Mind you, it's not generally portable to reuse a TCP socket for multiple connections, or attempts.

We're underspecified here, but we do generally try to be POSIX-ish, and from http://pubs.opengroup.org/onlinepubs/009695399/functions/connect.html

> If connect() fails, the state of the socket is unspecified. Conforming applications should close the file descriptor and create a new socket before attempting to reconnect.

So I would say you should be doing a close/open pair first before retrying. Will be more likely to work on the variety of implementations of TCPSocket you could be hitting with lwIP/Nanostack/various off-chip network stacks (eg wifi/cellular modules).

kegilbert, 2018-04-05:
My bad, forgot to specify the stack. This was over ethernet on a K64F (LWIP). I'll try recreating the socket, thanks!

geky, 2018-04-05:
Sorry @kegilbert, I have quite a bit of backlog so I probably can't look into this for a while. @kjbracey-arm is much more knowledgeable about the network stack these days, so I'll leave it up to the two of you to figure out.

kegilbert, 2018-04-05:
Closing the issue, closing out the socket before retrying worked.

closed 2018-04-05 by kegilbert