ARMmbed/mbed-os #6543: Build error occurs at SysTimer file when defining LPTICKER on Cortex-A

TomoYamanaka, 2018-04-05:
### Description
Currently, Renesas are implementing **Low Power Ticker** feature that was introduced at last week's workshop.
Our mbed board's H/W incorporates Cortex-A, however, we faced the below build error because it seem that `SysTimer.cpp` was based on only Cortex-M.
Refs are here:
[Ref 1](https://github.com/ARMmbed/mbed-os/blob/feature-hal-spec-ticker/rtos/TARGET_CORTEX/SysTimer.cpp#L27), [Ref 2](https://github.com/ARMmbed/mbed-os/blob/feature-hal-spec-ticker/rtos/TARGET_CORTEX/SysTimer.cpp#L108)

Could you please provide `SysTimer.cpp` that can work on not only Cortex-M but Cortex-A?

- Type: Bug 
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
RZ_A1H | GR_LYCHEE

**Toolchain:**
GCC_ARM|ARM|IAR

**Expected behavior**
Build success

**Actual behavior**
Build error
Err message is here:
```
        [Error] rtx_core_cm.h@626,22: 'SCB' was not declared in this scope
        [Error] rtx_core_cm.h@626,35: 'SCB_ICSR_PENDSVSET_Msk' was not declared in this scope
        [Error] rtx_core_cm.h@631,3: 'SCB' was not declared in this scope
        [Error] rtx_core_cm.h@631,15: 'SCB_ICSR_PENDSVCLR_Msk' was not declared in this scope
        [Error] rtx_core_cm.h@636,3: 'SCB' was not declared in this scope
        [Error] rtx_core_cm.h@636,15: 'SCB_ICSR_PENDSVSET_Msk' was not declared in this scope
        [Error] rtx_core_ca.h@362,22: redefinition of 'void SVC_Initialize()'
        [Error] rtx_core_ca.h@367,25: redefinition of 'uint8_t GetPendSV()'
        [Error] rtx_core_ca.h@372,22: redefinition of 'void ClrPendSV()'
        [Error] rtx_core_ca.h@377,22: redefinition of 'void SetPendSV()'
        [Error] SysTimer.cpp@108,5: 'SCB' was not declared in this scope
        [Error] SysTimer.cpp@108,17: 'SCB_ICSR_PENDSTSET_Msk' was not declared in this scope

```

0xc0170, 2018-04-05:
@c1728p9 @bulislaw Please review

TomoYamanaka, 2018-04-12:
How is going ?

bulislaw, 2018-04-12:
@TomoYamanaka can Renesas contribute changes to SysTimer to enable it on cortex a?

TomoYamanaka, 2018-04-13:
@bulislaw 
Unfortunately, I don't know how to change to `SysTimer` to enable on Cortex-A.

MarceloSalazar, 2018-04-19:
Internal reference: IOTMORF-2317