ARMmbed/mbed-os #6548: ARMC6 fails to link on DISCO_L475VG_IOT01A_ISM43362 and NUCLEO_L476RG

teetak01, 2018-04-05:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_L476RG | DISCO_L475VG_IOT01A_ISM43362 

**Toolchain:**
ARMC6

**mbed-os sha:**
5.8.1

build mbed-os-example-client:

mbed compile -m NUCLEO_L476RG -t ARMC6

```
Link: mbed-os-example-client
Warning: L3912W: Option 'legacyalign' is deprecated.
Warning: L3912W: Option 'no_strict_wchar_size' is deprecated.
Warning: L3912W: Option 'no_strict_enum_size' is deprecated.
Warning: L6433W: Ignoring incompatible enum size attribute on object iostream.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object cxa_virtual.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object memory.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object locale.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object ios.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object abort_message.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object private_typeinfo.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object exception.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object new.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object string.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object system_error.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object mutex.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object exception.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object stdexcept.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object typeinfo.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object stdexcept.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object cxa_handlers.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object cxa_noexception.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object cxa_default_handlers.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object cxa_exception_storage.cpp.o.
Warning: L6439W: Multiply defined Global Symbol SimpleSpirit1::enable_spirit_irq() defined in invalid_group(SimpleSpirit1.o) rejected in favor of Symbol defined in .text._ZN13SimpleSpirit117enable_spirit_irqEv(radio_spi.o).
Warning: L6439W: Multiply defined Global Symbol mbed::Callback<void(int)>::call(int) const defined in invalid_group(SimpleSpirit1.o) rejected in favor of Symbol defined in .text._ZNK4mbed8CallbackIFviEE4callEi(SerialBase.o).
Warning: L6439W: Multiply defined Global Symbol mbed::Callback<void()>::call() const defined in invalid_group(ATParser.o) rejected in favor of Symbol defined in .text._ZNK4mbed8CallbackIFvvEE4callEv(SerialBase.o).
Warning: L6439W: Multiply defined Global Symbol void SPWFSAxx::attach<SpwfSAInterface, void(SpwfSAInterface::*)()>(T1*, T2) defined in invalid_group(SPWFSAxx.o) rejected in favor of Symbol defined in .text._ZN8SPWFSAxx6attachI15SpwfSAInterfaceMS1_FvvEEEvPT_T0_(SpwfSAInterface.o).
Warning: L6439W: Multiply defined Global Symbol std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>::basic_string__sub_object(const std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>&, unsigned, unsigned, const std::__1::allocator<char>&) defined in .text._ZNSt3__112basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEEC2ERKS5_jjRKS4_(string.cpp.o) rejected in favor of Symbol defined in .text._ZNSt3__112basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEEC2ERKS5_jjRKS4_(WNC14A2AInterface.o).
Warning: L6439W: Multiply defined Global Symbol std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>::~basic_string__sub_object() defined in .text._ZNSt3__112basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEED2Ev(string.cpp.o) rejected in favor of Symbol defined in .text._ZNSt3__112basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEED2Ev(WNC14A2AInterface.o).
Warning: L6439W: Multiply defined Global Symbol mbed::DirHandle::readdir()::ent defined in invalid_group(Dir.o) rejected in favor of Symbol defined in .bss._ZZN4mbed9DirHandle7readdirEvE3ent(FileSystem.o).
Error: L6218E: Undefined symbol __HeapLimit (referred from mbed_boot.o).
Error: L6218E: Undefined symbol __StackLimit (referred from mbed_boot.o).
Error: L6218E: Undefined symbol __StackTop (referred from mbed_boot.o).
Error: L6218E: Undefined symbol __end__ (referred from mbed_boot.o).
Error: L6218E: Undefined symbol __mbed_krbs_start (referred from l4_retarget.o).
Error: L6218E: Undefined symbol __mbed_sbrk_start (referred from l4_retarget.o).
Finished: 0 information, 30 warning and 6 error messages.
[ERROR] Warning: L3912W: Option 'legacyalign' is deprecated.
Warning: L3912W: Option 'no_strict_wchar_size' is deprecated.
Warning: L3912W: Option 'no_strict_enum_size' is deprecated.
Warning: L6433W: Ignoring incompatible enum size attribute on object iostream.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object cxa_virtual.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object memory.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object locale.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object ios.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object abort_message.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object private_typeinfo.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object exception.cpp.o.
[... 3 lines cut ...]
Warning: L6433W: Ignoring incompatible enum size attribute on object mutex.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object exception.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object stdexcept.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object typeinfo.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object stdexcept.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object cxa_handlers.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object cxa_noexception.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object cxa_default_handlers.cpp.o.
Warning: L6433W: Ignoring incompatible enum size attribute on object cxa_exception_storage.cpp.o.
Warning: L6439W: Multiply defined Global Symbol SimpleSpirit1::enable_spirit_irq() defined in invalid_group(SimpleSpirit1.o) rejected in favor of Symbol defined in .text._ZN13SimpleSpirit117enable_spirit_irqEv(radio_spi.o).
Warning: L6439W: Multiply defined Global Symbol mbed::Callback<void(int)>::call(int) const defined in invalid_group(SimpleSpirit1.o) rejected in favor of Symbol defined in .text._ZNK4mbed8CallbackIFviEE4callEi(SerialBase.o).
Warning: L6439W: Multiply defined Global Symbol mbed::Callback<void()>::call() const defined in invalid_group(ATParser.o) rejected in favor of Symbol defined in .text._ZNK4mbed8CallbackIFvvEE4callEv(SerialBase.o).
Warning: L6439W: Multiply defined Global Symbol void SPWFSAxx::attach<SpwfSAInterface, void(SpwfSAInterface::*)()>(T1*, T2) defined in invalid_group(SPWFSAxx.o) rejected in favor of Symbol defined in .text._ZN8SPWFSAxx6attachI15SpwfSAInterfaceMS1_FvvEEEvPT_T0_(SpwfSAInterface.o).
Warning: L6439W: Multiply defined Global Symbol std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>::basic_string__sub_object(const std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>&, unsigned, unsigned, const std::__1::allocator<char>&) defined in .text._ZNSt3__112basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEEC2ERKS5_jjRKS4_(string.cpp.o) rejected in favor of Symbol defined in .text._ZNSt3__112basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEEC2ERKS5_jjRKS4_(WNC14A2AInterface.o).
Warning: L6439W: Multiply defined Global Symbol std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>::~basic_string__sub_object() defined in .text._ZNSt3__112basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEED2Ev(string.cpp.o) rejected in favor of Symbol defined in .text._ZNSt3__112basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEED2Ev(WNC14A2AInterface.o).
[... cut ...]
Error: L6218E: Undefined symbol __HeapLimit (referred from mbed_boot.o).
Error: L6218E: Undefined symbol __StackLimit (referred from mbed_boot.o).
Error: L6218E: Undefined symbol __StackTop (referred from mbed_boot.o).
Error: L6218E: Undefined symbol __end__ (referred from mbed_boot.o).
Error: L6218E: Undefined symbol __mbed_krbs_start (referred from l4_retarget.o).
Error: L6218E: Undefined symbol __mbed_sbrk_start (referred from l4_retarget.o).
Finished: 0 information, 30 warning and 6 error messages.

```

0xc0170, 2018-04-05:
@adustm Hi, this as I have reviewed the above issue is related to TWO RAM REGIONS added. 
Note this is reported for ARMC6 that is currently not supported on master. Not sure if actually a build should fail differently than missing some symbols (is not caught anywhere).

@teetak01 Please confirm, removing the TWO RAM regions commit fixes the above issue

jeromecoutant, 2018-04-17:
This could be closed thx to #6553 ?

closed 2018-04-17 by 0xc0170