ARMmbed/mbed-os #6549: STM32 External Crystal configured for board that reports external clock support

pauluap, 2018-04-05:
### Description
- Type:  Question

**Target**
DISCO_F769NI

**Toolchain:**
GCC_ARM

**Toolchain version:**
(GNU Tools for ARM Embedded Processors 6-2017-q1-update) 6.3.1 20170215 (release) [ARM/embedded-6-branch revision 245512]

I'm looking at the clock configuration, and I'm trying to understand an apparent configuration mismatch with the physical board.

Looking at the DISCO-F769 schematics (one location is the DISCO-F769 user manual, http://www.st.com/content/ccc/resource/technical/document/user_manual/group0/7c/c9/c0/5f/4e/9b/44/eb/DM00276557/files/DM00276557.pdf/jcr:content/translations/en.DM00276557.pdf, page 41, it reports that an external clock is used rather than an external crystal.  Observing my board, it appears to match the schematics.

However, the mbed-config.h file has the clock configured as
`#define CLOCK_SOURCE                                      USE_PLL_HSE_XTAL|USE_PLL_HSI // set by target:DISCO_F769NI`

Rather than
`#define CLOCK_SOURCE                                      USE_PLL_HSE_EXTC|USE_PLL_HSI // set by target:DISCO_F769NI`

Is the sense of `USE_PLL_HSE_XTAL` and `USE_PLL_HSE_EXTC` flipped?  That doesn't appear likely from looking at the source code

https://github.com/ARMmbed/mbed-os/blob/16bac101a6b7b4724023dcf86ece1548e3a23cbf/targets/TARGET_STM/TARGET_STM32F7/TARGET_STM32F769xI/TARGET_DISCO_F769NI/system_clock.c#L153

Is it that a external clock are usable even when the clock system is configured for a crystal?

bcostm, 2018-04-06:
I agree with you that we should put USE_PLL_HSE_EXTC in the targets.json when the clock is coming from an external oscillator on pin OSCIN. It is a mistake in the targets.json. But I can't tell why it is working like it is today. Maybe the clock system in this MCU doesn't care ? I have re-checked both configs (XTAL and EXTC) and both are working fine. I will send a PR to correct the targets.json to clarify it. Thanks.

jeromecoutant, 2018-04-25:
I think we can close this issue ?

closed 2018-04-25 by 0xc0170