ARMmbed/mbed-os #6554: Unable to send ”Hello world" from HC-06 Bluetooth with K64f mbed to Android phone terminal app

Robinmarsx, 2018-04-05:
[image]
As the picture shows, please help me out. Cannot send text from HC-06 with freescale K64F board to Android phone terminal App.

0xc0170, 2018-04-06:
@Robinmarsx Can you provide more information? The code snippet is fine, but rather than picture something that can be copy-paste ? What version of mbed OS are you using? HC06 - library? As much details as you can

Unable to send - does it fail (any IRQ handler) or just executes, nothing to be sent?

From the snippet, looks like you are using serial on those 2 pins, are they connected to a sensor or ?

0xc0170, 2018-05-22:
Closing as no update.

closed 2018-05-22 by 0xc0170