ARMmbed/mbed-os #6556: LoRaPHY.cpp CFList decode error.

tpet93, 2018-04-06:
### Description
- Type: Bug 
- Priority:  Major 

---------------------------------------------------------------
## Bug
Regions with cflist_supported == true
send the channel frequency in groups of 3 bytes per channel.

In the function LoRaPHY::apply_cf_list 
The byte position is incremented by "phy_params.default_channel_cnt" when it should be 3.

Current
```c++
    for (uint8_t i = 0, channel_id = phy_params.default_channel_cnt;
         channel_id < phy_params.max_channel_cnt; i+=phy_params.default_channel_cnt, channel_id++) {
        if (channel_id < (phy_params.cflist_channel_cnt + phy_params.default_channel_cnt)) {
            // Channel frequency
            new_channel.frequency = (uint32_t) cf_list->payload[i];
            new_channel.frequency |= ((uint32_t) cf_list->payload[i + 1] << 8);
            new_channel.frequency |= ((uint32_t) cf_list->payload[i + 2] << 16);
            new_channel.frequency *= 100;
```
Fixed
```c++
    for (uint8_t i = 0, channel_id = phy_params.default_channel_cnt;
         channel_id < phy_params.max_channel_cnt; i+=3, channel_id++) {
        if (channel_id < (phy_params.cflist_channel_cnt + phy_params.default_channel_cnt)) {
            // Channel frequency
            new_channel.frequency = (uint32_t) cf_list->payload[i];
            new_channel.frequency |= ((uint32_t) cf_list->payload[i + 1] << 8);
            new_channel.frequency |= ((uint32_t) cf_list->payload[i + 2] << 16);
            new_channel.frequency *= 100;
```

some regions send a CFList as channel masks with 2 bytes per block of 16 Channels (AU915-928)
however this looks to be require adding additional code to the apply_cf_list function.
theses regions have the cflist_supported flag set to false so the above fix will not break any functionality

**Expected behavior**
turn 3 bytes into one uint32_t

**Actual behavior**
depending on regions (as923 for example) only uses 2 bytes into one uint32_t

----------------------------------------------------------------

0xc0170, 2018-04-06:
@AnttiKauppila @kivaisan  Please review

AnttiKauppila, 2018-04-06:
ARM internal ref: IOTCELL-754

AnttiKauppila, 2018-04-06:
This is a real issue in our stack and it was well spotted.
We will check other regions (2 byte) if those need any modifications.
Fix done here as suggested by tpet93:
https://github.com/ARMmbed/mbed-os/pull/6557

closed 2018-04-09 by 0xc0170