ARMmbed/mbed-os #6578: LPC1768, hard fault when calling fopen() after Ethernet connect

janjongboom, 2018-04-09:
### Description
- Type: Bug

---------------------------------------------------------------
## Bug

**Target**
LPC1768

**Toolchain:**
ARM

**mbed-os sha:**
Mbed OS 5.4 - 5.8, Mbed OS 5.3 and Mbed 2 not affected.

**Expected behavior**

Calling `fopen()` after calling `connect` on the ethernet interface does not throw an error.

**Actual behavior**

It hard faults.

**Steps to reproduce**

Example program is here https://os.mbed.com/questions/80658/HardFault-occurs-when-fopen-is-called-af/

User verified working behavior on Mbed 2 and Mbed OS 5.3; broken behavior on 5.4 and 5.8.

0xc0170, 2018-04-09:
@kjbracey-arm @mikaleppanen Please review

kjbracey-arm, 2018-04-10:
I'm reckoning the exception is happening as `__semihost(SYS_OPEN, args)` is called from `semihost_open` (R0=1=`SYS_OPEN`, R1=current thread stack pointer).

`HFSR` (Hard Fault Status Register) has bit 31 set - `DEBUGEVT`. The `BKPT` instruction that's suppose to be captured by the semihosting hasn't been, so it just generates a hard fault.

Somehow starting the Ethernet has stopped the semihosting from working - deactivated the debug connection or made it lose sync?

Now, I know nothing much about the semihosting implementation - I suggest referring this to someone who knows how that's supposed to work, and what could interfere with it.

Maybe someone who can reproduce it could try a `git bisect` on mbed OS - might be the fastest way to isolate. I can't obviously see anything touching LPC1768 Ethernet or debug/semihosting in between 5.3 and 5.4.

kjbracey-arm, 2018-04-13:
Looks like it's nothing to do with the Ethernet - same basic effect can be triggered much more simply as shown in #6624.

maclobdell, 2018-06-12:
cc @mmahadevan108

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-6047*

Nuclearfossil, 2019-08-13:
Is this actually resolved? I'm seeing this exact issue when sending code to an FPGA across the SPI.

I'm migrating code from OS 2 to 5.13 and I'm seeing hard faults whenever I inject a `wait_ms`.

For example:

```
void customFPGALoader()
{
    SPI spi(p5, p6, p7);
    DigitalOut cs(p8)
    , led1(LED1)
    , led2(LED2)
    , led3(LED3)
    , led4(LED4)
    , spi_en(p17);

    LocalFileSystem local("local");

    led1 = 1;
    cs = 0;
    spi.format(8);
    spi.frequency(1000000);

    spi_en = 1;
    wait_ms(50);

    const uint32_t FPGAMAXIMAGE = 40000;
    FILE* pfp = fopen("/local/FPGA.img", "rb");

    if(pfp != nullptr)
    {
        for(uint32_t i = 0; i < FPGAMAXIMAGE && feof(pfp); i++)
        {
            spi.write(fgetc(pfp));
        }
        fclose(pfp);

        for(uint32_t i = 0; i < 10;i++)
        {
            spi.write(0xFF);
        }
    }

    cs = 1;
    spi.frequency(10000000);
    spi_en = 1;
    led2 = 1;
}
```

kjbracey-arm, 2019-08-14:
I've not heard of any progress.

Since this issue was filed, I've become a bit more familiar with the `LocalFileSystem` stuff - used it on non-LPC1768 devices to get semihosted console, as seen now in an example here: https://os.mbed.com/docs/mbed-os/v5.13/apis/filehandle.html.

Looking at the LPC1768 sleep implementation, we see this:

https://github.com/ARMmbed/mbed-os/blob/b82cfcc38be1b93fdb7a7192100569f060011569/targets/TARGET_NXP/TARGET_LPC176X/sleep.c#L27-L32

That's what causes the issue - the moment the device sleeps (because of the `wait_ms`), the debug connection that `LocalFileSystem` for its semihosting calls is using is dropped.

The behaviour is explicitly documented:

https://github.com/ARMmbed/mbed-os/blob/b82cfcc38be1b93fdb7a7192100569f060011569/targets/TARGET_NXP/TARGET_LPC176X/sleep.c#L21-L24

Comments elsewhere in that file suggest that the LPC1768 actually has issues with losing debug in deep sleep (as do most devices), so I wonder whether the disconnect is really necessary for a _normal_ sleep.

The usual setup in other devices is that `hal_sleep` does not disrupt the debug port, and `hal_deepsleep` does.

A debug profile build globally stops the sleep manager from ever using `hal_deepsleep`, avoiding debug problems.

Maybe `LocalFileSystem` itself should be locking deep sleep while active, but it doesn't. But in LPC1768's case, both sleep implementations are the same anyway.

It would be good if someone at @ARMmbed/team-nxp could look at this.

In the meantime, maybe you could just experiment and take out that disconnect call? Have a look at LPC1768 documentation for hints about the way the sleeps should be working?

Nuclearfossil, 2019-08-14:
Hey Kevin,

Thanks for the feedback and heads up.

Interesting - I noticed that there's no 'profile' setting in the online compiler (that I could see) for picking release/developer/debug. So that could be the overall issue in my case; online compiled versions of the codebase did not hardfault, whereas mbed studio compiled versions did.

I may just see about doing a developer/release build and see if the issue persists.

I'm hesitant on removing the `wait_ms` calls in the original codebase as I'm not sure what the intent was (original developer has left, no documentation, the standard story on this stuff).

Again, thanks.

adbridge, 2020-10-13:
We've updated our automation, I will fix the requirements .

defcon8, 2022-03-05:
We face the same issue. Accessing mBed OS5 LocalFileSystem after SPI bus communication took place results in a hardfault.

Any update on this issue? We need to be able to write some settings to the flash rom after SPI is initialized.

0xc0170, 2022-03-07:
@ARMmbed/team-nxp would you be able to help?

@defcon8 the reply from Kevin above provides good details related to sleep and wait_ms. Or just use storage instead of localfilesystem?