ARMmbed/mbed-os #6581: custom_targets.json and modular target board support packages

loverdeg-ep, 2018-04-09:
### Description
- Type: Question, possible enhancement
- Priority: Major

---------------------------------------------------------------
@theotherjimmy @0xc0170 

## Background

Assume I create a custom target and want to use that target in multiple applications.
The most elegant approach I can think of with my limited understanding of the config system, would be to create a git repository that contains the following:

* PinNames.h
* device.h
* custom_targets.json
* git submodules of any of the board's IC drivers not already included with mbedOS

This repository would then be imported as a git submodule of any applications I develop for the board.

(or, if the application is hardware agnostic enough, I might create a new repository that includes both the application and BSP as  git submodules)

Known Issues to using this approach:
* custom_targets.json needs to be in the root of the "project"
* managing mbedOS version dependencies 

## Questions

* What currently is the cleanest way to accomplish modular custom target support for an application?

* What, if anything, could be done to 'enhance' custom target support?

theotherjimmy, 2018-04-09:
@loverdeg-ep The current limitation, that the "custom_targets.json" must be at the root, is primarily an implementation limitation. You could "tick" the build system into thinking that it's at the root by:
 * Place an `.mbedignore` file in the root ignoring the module directory
 * always run `mbed compile`/`mbed export` with `--source` pointing to that dir.

I take it that this is almost a blocker for going to production?

loverdeg-ep, 2018-04-09:
@theotherjimmy Not quite a blocker since we can work around it.
Just want to make sure custom_targets get first class support moving forward.

theotherjimmy, 2018-04-09:
@loverdeg-ep Yeah, It seems that custom target support is really needed for going to production, as you use it for the custom board you're making. I appreciate that you're pushing on it, as it frequently sees the lest love.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-323*

theotherjimmy, 2019-01-15:
@loverdeg-ep The implementation that you outlined would work, and I would accept a PR that adds and implements a `--custom-targets` command line switch.

theotherjimmy, 2019-01-15:
> Can't put my finger on it, but using custom_targets.json to define custom targets feels ick in the first place.

Good to know that something feels off to you. If you can put your finger on what that is, I would like to know it.

evedon, 2019-05-24:
--custom-targets command line switch was delivered in #9401 so closing this issue.

closed 2019-05-24 by evedon