ARMmbed/mbed-os #6585: CMake exported build fails with multiple --source arguments

vidavidorra, 2018-04-10:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Related issue: #6143 and ARMmbed/mbed-cli#590
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
All targets (e.g. `TARGET_NUCLEO_F103RB`)

**Toolchain:**
GCC_ARM

**Toolchain version:**
7.2.1

**mbed-cli version:** 
1.5.0

**mbed-os sha:**
192a8e509

---------------------------------------------------------------
Folder structure I use for my project:
```
├── project.cmake
├── src
│   └── main.cpp
└── submodules
    ├── mbed-memory-status
    └── mbed-os
```

**Expected behavior**
```
mbed config MBED_OS_DIR submodules/mbed-os
mbed export -i make_gcc_arm -m NUCLEO_F103RB --source src --source submodules --build .
make -j
```

This generates the file `src.elf` since it automatically names the project what I give as the first `--source`. This builds successfully.

**Actual behavior**

```
mbed config MBED_OS_DIR submodules/mbed-os
mbed export -i cmake_gcc_arm -m NUCLEO_F103RB --source src --source submodules --build .
cmake .
make all

...

[100%] Linking CXX executable src
/usr/local/Cellar/gcc-arm-none-eabi-74/20171218/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: cannot open output file src: Is a directory
collect2: error: ld returned 1 exit status
```
Tries to generate the file `src` which results in an error in the linker since it has the same name as my `src` folder with no extension. In my opinion this file should get the `elf` extension, which would resolve the problem. I also expect to be able to specify the project name to the export command, which at the moment is not possible according to the mbed-cli [arguments](https://github.com/ARMmbed/mbed-cli/blob/master/mbed/mbed.py#L2578-L2585) to the export tool.

**Steps to reproduce**

----------------------------------------------------------------
## Enhancement
The workaround I found to do mostly what I want is to add a line in `project.cmake` that changes the `OUTPUT_NAME` of the target property to the name specified as the first `--source` argument.

```
set_target_properties(src PROPERTIES OUTPUT_NAME "mbed-os-example")
```
but as said, that is merely a workaround

**Reason to enhance or problem with existing solution**
Build (linker) files on CMake export with multiple `--source` arguments

**Suggested enhancement**
I suggest adding a argument to the mbed-cli (and other parts of the export) to specify the project name e.g. `-p mbed-os-example` or `--project mbed-os-example` that adds a line to the `CMakeLists.txt` to specify the output name. Ideally the output extension for the CMake export should be changed to `elf` like the `make_gcc_arm` export does.

theotherjimmy, 2018-04-19:
@vidavidorra the project name may be set the `mbed_app.json` section `artifact_name` ([docs](https://os.mbed.com/docs/v5.8/reference/configuration.html)). If it's not set, it defaults to the name of the directory of the first `--source` argument, which was probably your `src` dir. I agree that the cmake exporter should probably append a `.elf`.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-324*

vidavidorra, 2018-10-12:
@theotherjimmy I've updated to mbed-os-5.10.0 and tested this again, especially specifying the name using the `artifact_name` you mentioned. I've also tested `artifact-name` but with same outcome.

```
{
  "config": {
    "artifact_name" : "example",
...
```

I'm using that with cmake exporter
```
mbed export -i cmake_gcc_arm --app-config mbed_app.json --source ./source
```

but when I look in the generated `CMakeLists.txt` I see that `source` is still used as name. I.e. `PROJECT(source C CXX ASM)` is in there where I'd now expect example instead of source. This is not only for project name but also for `SET_TARGET_PROPERTIES`, `add_custom_command(TARGET` etc.

vidavidorra, 2018-10-12:
@theotherjimmy I've also just tested the CMake export on mbed-os-5.10.0 and don't think #8251 is necessary anymore. Not sure why though, but the linker no longer gives the error.

Also I'm wondering what the project name is for other exports (GCC_ARM, Make). If those end with .elf I think #8251 still makes sense, but if they don't IMHO #8251 is not necessary.
If the latter is the case IMHO the elf extension can simply be added to the output file using the following rather than renaming the project. That is, if there is not a more elegant way to do so and this doesn't break anything else.
```cmake
add_custom_command(TARGET source POST_BUILD
    COMMAND ${CMAKE_COMMAND} -E rename $<TARGET_FILE:source> $<TARGET_FILE:source>.elf
    )
```

closed 2019-02-05 by ciarmcom