ARMmbed/mbed-os #6589: Add compiler in test result summary

bcostm, 2018-04-10:
### Description
- Type: Enhancement
- Priority: Minor

----------------------------------------------------------------
## Enhancement

At the end of the generated test report (mbed5 or mbed2) there is mention of the number of FAIL/OK/TIMEOUT/SKIPPED... like this:

`6 FAIL / 56 SKIPPED / 1450 OK / 3 ERROR` for mbed5
or
`Result: 3 OK/ 2 FAIL` for mbed2

It would be interesting to detail this summary per compiler.

Something like for mbed5:

```
ARM: 2 FAIL / 10 OK / 3 TIMEOUT
GCC_ARM: 10 OK / 5 TIMEOUT
IAR: 15 OK
```
and for mbed2:
`Result ARM: 3 OK / 2 FAIL`
`Result GCC_ARM: 5 OK`

0xc0170, 2018-04-10:
@theotherjimmy @MarceloSalazar please review

theotherjimmy, 2018-04-10:
@bcostm Could you be specific about what command you run?

bcostm, 2018-04-11:
for example:
`mbed test -vv --run -m NUCLEO_L073RZ -t ARM --report-text=report_mbed5.txt`
or
`./tools/singletest.py --auto -f NUCLEO_L073RZ --tc ARM -j 4 -c copy -V -v -n MBED_A1 --report-text=report_mbed2.txt`

But the tests summary is also displayed on the screen even without the `--report-text` option.

MarceloSalazar, 2018-04-13:
This info seems very useful :+1:

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-325*

closed 2019-10-31 by ciarmcom