ARMmbed/mbed-os #6591: lp_ticker and RTC tests failing for platforms based on STM32F412xx

ashok-rao, 2018-04-10:
### Description
- Type: Bug | Question
- Related PR's: #6511 and #6423 
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
MTB_USI_BN-BM-22 and MTB_ADV_WISE_1530

**Toolchain:**
ARM

**Expected behavior**
All greentea tests should pass for the targets.

**Actual behavior**
LP_Ticker, LP_timer and RTC tests are failing on ARMCC 5.06 update 3.

**Steps to reproduce**
Run greentea on these 2 targets with ARMCC

**Greentea logs are attached in the PR**: https://github.com/ARMmbed/mbed-os/pull/6511#issue-178369355 
And failing tests for ADV_WISE_1530 are added as comments here: https://github.com/ARMmbed/mbed-os/pull/6423#issuecomment-377179117

jeromecoutant, 2018-04-11:
Hi

I know that NUCLEO_F412ZG is failing tests-mbed_drivers-lp_timeout since 5.8.1
Other LP tests are OK.
No known issue with RTC tests.

Note that we will not investigate on LP failed tests before the ticker branch merge.

jeromecoutant, 2018-04-11:
We can discuss in parallel about your clock configuration:

- USI WM-BN-BM-22: I can see from the targets json file that you have a LSE and no HSE.
This is OK, but could be a problem for tests part... as several tests check the accuracy of time with the us timer which is clocked in your case with HSI, the internal low cost clock.
As you have an external LSE clock, idea could be to make a calibration procedure.

- MTB_ADV_WISE_1530: targets json file doesn't explicit your clock configuration, which is not correct.

0xc0170, 2018-04-11:
> Note that we will not investigate on LP failed tests before the ticker branch merge.

You are referring to the ticker hal specs feature branch? This then becomes known issue for 5.8.x - should these target have disabled low ticker as it is not fully functional?

0xc0170, 2018-04-11:
Looking at 5.8.1 - https://github.com/ARMmbed/mbed-os/pull/6461 - tests were updated (timeout) - that would be the reason for the failures @fkjagodzinski Please confirm

fkjagodzinski, 2018-04-11:
That is correct, #5106 which contains an update for LowPowerTimeout  tests has been merged to 5.8.1.

0xc0170, 2018-04-13:
so that update broke it. @fkjagodzinski Can you have look ? Is it a test or targets?

fkjagodzinski, 2018-04-17:
Yes, these tests are built only when `LOWPOWERTIMER` macro is defined, so you should either edit targets.json or use `-D` compile option.

ashok-rao, 2018-04-17:
@fkjagodzinski But, we did not have LOWPOWERTIMER defined in targets.json for the commit 1df5c020b966ac464edde0748903b0693de3aa2f  too. How come it was being built then..!

fkjagodzinski, 2018-04-17:
@ashok-rao No idea.. :confused: Morph build logs from https://github.com/ARMmbed/mbed-os/pull/6511#issuecomment-378182965, e.g. [[...]build_log_MTB_USI_WM_BN_BM_22_GCC_ARM.txt](http://mbed-os.s3-eu-west-1.amazonaws.com/builds/6511/PASS/MTB_USI_WM_BN_BM_22/GCC_ARM/62e5dccd0b28738ef4b2bbf0ecacdcac17284d4c_build_log_MTB_USI_WM_BN_BM_22_GCC_ARM.txt) list these tests as skipped.

KariHaapalehto, 2018-05-04:
RTC tests will pass by setting "overrides": {"lse_available": 0} to USI_WM_BN_BM_22
Who can verify is it ok to disable lse?

jeromecoutant, 2018-05-04:
> Who can verify is it ok to disable lse?

Ask your HW experts :-)
Is a 32k clock is provided in the board ?
Are OSC32 pins connected ?

kivaisan, 2018-07-06:
According to MCB USI_WM_BN_BM_22 schematics 32KHZ_IN pin is not connected so I think then "lse_available": 0 is ok. Right?

0xc0170, 2018-08-01:
@ashok-rao Is this resolved, or any action still to be done?

ashok-rao, 2018-08-01:
@0xc0170 : Resolved by PR #7446 . Thank you. Closing the issue now.

closed 2018-08-01 by ashok-rao