ARMmbed/mbed-os #6593: How does the build system work with regard to targets?

loverdeg-ep, 2018-04-10:
### Description
- Type: Question
- Priority: Minor

## Question

I think I can boil this one down to:

How does the python based 'build' system under the tools directory work in regards to finding target specific sources, linker scripts, startup code etc?

## Notes
My question is arises from having a customer who might want to provide their own linker file as an application level configuration for a custom target based on one of the nrf parts.

We have found that the -l option to mbed-cli when compiling to specify a linker script. I assume the -l gets passed on to make.py; however this option doesn't seem to be documented anywhere.

0xc0170, 2018-04-11:
@theotherjimmy Please review

loverdeg-ep, 2018-04-11:
So the system enters that directory and recursively follows paths looking for sources, linker scripts, startup code, etc?

Is it looking at cmsis-packs for the mcu's at all?

theotherjimmy, 2018-04-11:
> So the system enters that directory and recursively follows paths looking for sources, linker scripts, startup code, etc?

Yes

>Is it looking at cmsis-packs for the mcu's at all?

Only if they have a `device_name` and then only for ram/rom sizes and locations.

closed 2018-04-24 by loverdeg-ep