ARMmbed/mbed-os #6594: LoRa Phy AU915 channel parameters not being used, causing Up-links at 500khz

tpet93, 2018-04-11:
### Description
- Type: Bug 
- Priority:  Major

---------------------------------------------------------------
## Bug

I was having issues with the min and max data rates for each channel for the AU915 Region.

In the function ```LoRaPHY::enabled_channel_count```,
The values returned by 
```phy_params.channels.channel_list[i].dr_range.fields.max``` and ```phy_params.channels.channel_list[i].dr_range.fields.min```
appear to be semi-random as if not initialized.

causing the valid channels for a specific data-rate to return unexpected values.

This frequently causes the up-links to be sent on one of the 500khz channels (64-71).

Most other regions include the block of code below in LoRaPHYXXXXX.cpp
around line 255 above the line     ```phy_params.bands.table = (void *) bands;```

```
    // set default channels
    phy_params.channels.channel_list = channels;
    phy_params.channels.channel_list_size = AU915_MAX_NB_CHANNELS;
    phy_params.channels.mask = channel_mask;
    phy_params.channels.default_mask = default_channel_mask;
    phy_params.channels.mask_size = AU915_CHANNEL_MASK_SIZE;
```

adding the above block of code into LoRaPHYAU915.cpp around line 266 appears to fix the issue.

-----------------------------------------------------------------

kivaisan, 2018-04-11:
@hasnainvirk, please check this.

hasnainvirk, 2018-04-11:
It's legit. Will make an internal issue about it.

hasnainvirk, 2018-04-11:
Issue being tracked here https://jira.arm.com/browse/IOTCELL-761

hasnainvirk, 2018-04-11:
@tpet93 looking at the LoRaWAN Regional Parameters1.0.2Final_1944_1 document, our code looks correct. Please refer to the line number 779 to 782. 

> If using the over-the-air activation procedure, the end-device should broadcast the JoinReq message alternatively on a random 125 kHz channel amongst the 64 channels defined using DR0 and a random 500 kHz channel amongst the 8 channels defined using DR6. The end-device should change channel for every transmission.

Can you please refer me to any standard document where it proposes a solution which corresponds to the solution you are proposing here ?

hasnainvirk, 2018-04-11:
@tpet93 https://github.com/ARMmbed/mbed-os/pull/6600 PR open for the bug fix. Thanks for spotting it.

kivaisan, 2018-06-04:
@tpet93 Could you please check if this can be closed as fixed?

tpet93, 2018-06-21:
@kivaisan 
Yes the Issue described was fixed. Thanks

closed 2018-06-21 by 0xc0170