ARMmbed/mbed-os #6597: Reinforce usage of gcc v6 as gcc v5 doesn't work

MarceloSalazar, 2018-04-11:
### Description
- Type: Bug & need enhancement
- Priority:  Major

---------------------------------------------------------------
## Bug

During training, we've found out that some developers had gcc v5 installed and were able to compile/link and program the target. However, the app doesn't work.

**Target**
K64F and others

**Toolchain:**
GCC_ARM

**Toolchain version:**

gcc v5 --> problem
gcc v6 --> supported & works fine
https://os.mbed.com/docs/v5.8/tools/index.html

**mbed-cli version:** 
(`mbed --version`)

**mbed-os sha:**
(`git log -n1 --oneline`)

**Expected behavior**

When using gcc v5, the Mbed tools should display a message saying that toolchain is not supported.

**Actual behavior**

It silently compiles and links, giving no indication to the user that there is a problem with an unsupported toolchain that won't work.

0xc0170, 2018-04-11:
> During training, we've found out that some developers had gcc v5 installed and were able to compile/link and program the target. However, the app doesn't work.

This could be actually true for any compiler out there. As we progress, we update from time to time - thus tested only the current set of supported compilers. Would this make sense to extend to all and check the range if it's valid (assuming range is less strict as specific version, IAR 7.x for now should work for instance?)

Is this enhancement for this repo or mbed cli?

@theotherjimmy Suggestions?

MarceloSalazar, 2018-04-11:
> Is this enhancement for this repo or mbed cli?

This should be part and defined within Mbed OS -Mbed CLI can run with multiple versions of Mbed OS.

> Would this make sense to extend to all and check the range if it's valid

IMO this should be extended to all supported compilers as documentation states:
- ARM: v5 Update >=3 (v6 may work)
- GCC: v6.x (v7.x may work)
- IAR > 7.8 (v8.x may work)

screamerbg, 2018-04-11:
@0xc0170 This is enhancement for the Mbed OS build tools as with every Mbed OS release the toolchain/compiler version requirements will change.

cmonr, 2018-08-13:
Closing since https://github.com/ARMmbed/mbed-os/pull/7247 has been merged, and addresses the original request.

closed 2018-08-13 by cmonr