ARMmbed/mbed-os #6617: Request to incorporate the revision of RTX5 for Cortex-A

TomoYamanaka, 2018-04-12:
### Description
The problem that program abort occurs when run `Mbed Cloud Client` sample program on GR-PEACH / GR-LYCHEE.
As a result of our analysis, we faced unexpected behavior if interrupt is fired during `SVC` in RTX5.
There are three problem points in each `irq_ca.s` file, We already submitted the issue to Arm CMSIS_5 team and asked the revise to them.
For your information, it seems that this symptom occurs only Cortex-A.
The following commit that relating fix is incorporated into CMSIS-5 repo.
 https://github.com/ARM-software/CMSIS_5/commit/815dfadcec8465a9185a51bb1566bc40e08b7752
 https://github.com/ARM-software/CMSIS_5/commit/0b7a4a9f170f6693db98c1fab15318ec7fd28510

Since we confirmed that this multi interrupt issue was resolved, could you cherry-pick their commit to mbed-os master branch ?

- Type: Bug 
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
RZ_A1H | GR_LYCHEE

**Toolchain:**
GCC_ARM|ARM|IAR

0xc0170, 2018-04-12:
Would this update resolve this https://github.com/ARMmbed/mbed-os/pull/6273 ?

cc @bulislaw

TomoYamanaka, 2018-04-12:
I think that it will not be solved since the above commits is not listed in #6273.

Meano, 2018-04-16:
I tried to replace these files. It resolved this issue #6355 .

bulislaw, 2018-04-16:
We'll need to cherry pick the commits on top of the update after #6273 I'll try to resurrect it and merge soon.

TomoYamanaka, 2018-04-19:
@bulislaw 
Could you tell us the expected date of cherry pick two commits ?

TomoYamanaka, 2018-04-25:
Since I think that the above two commits will not be affected to #6273 , can I submit PR of cherry-pick to go ahead with the reflection to Mbed OS?
It seems like the multiple interrupts problem in RTX5 blocks GR-PEACH/GR-LYCHEE users' activity.
Thus, if you can accept my cherry-pick on this soon, it would be very helpful.

bulislaw, 2018-04-25:
I'll cherry pick that on top of https://github.com/ARMmbed/mbed-os/pull/6273

TomoYamanaka, 2018-05-21:
How is this going ?

TomoYamanaka, 2018-06-04:
I close because addressed this issue by #7032 .

closed 2018-06-04 by TomoYamanaka