ARMmbed/mbed-os #6623: SPI Clock Not Initialized Properly

AGlass0fMilk, 2018-04-12:
### Description
- Type: Bug/Question
- Priority: Minor

Not completely sure if this is an issue, but it caused me some head scratching. It can be fixed by the application by initializing the pin as a DigitalOut and manually setting the clock pin to the idle state. But I think this should be taken care of by the driver.

See attached image

[image]

---------------------------------------------------------------
## Bug

**Target**
Nordic nRF52832-based targets

**Toolchain:**
GCC_ARM

**Toolchain version:**

**mbed-cli version:** 
1.2.2

**Expected behavior**
SPI Clock pin should be initialized to idle state (depending on mode) when SPI driver is initialized.

**Actual behavior**
SPI clock pin is not initialized to idle state until first transaction starts, which could cause issues. It shows up on my Saleae logic analyzer as an incorrect setting/error

**Steps to reproduce**
Use SPI driver in mode 0 on nRF52832 targets and capture first transaction on a logic analyzer. The state of the clock pin will be incorrect.

0xc0170, 2018-04-13:
@marcuschangarm Please review

marcuschangarm, 2018-04-13:
@AGlass0fMilk 

* Is this with the new driver on the feature branch?
* What device (if any) are you testing against?

Thanks!

AGlass0fMilk, 2018-04-13:
@marcuschangarm 
- Yes, this is with the new driver on the feature branch. I just recently synced my development branch with the feature branch.

- I am testing on a custom device using an nRF52832

marcuschangarm, 2018-04-16:
I see the problem. Let us get https://github.com/ARMmbed/mbed-os/pull/6547 merged and I'll put up a bug fix.

marcuschangarm, 2018-04-23:
@AGlass0fMilk can you give this a spin and see if it solves your problem, please?

https://github.com/ARMmbed/mbed-os/pull/6717

All pins on my NRF52_DK board are low by default so I can't test it properly.

loverdeg-ep, 2018-06-13:
@AGlass0fMilk Can you confirm if this solved your problem?

AGlass0fMilk, 2018-06-20:
I will report back when I have a logic analyzer on my SPI next.

Seeing another with the SPI driver here (see #7278)

0xc0170, 2018-07-03:
This should be also resolved , same as the another referenced issue above

closed 2018-07-03 by 0xc0170