ARMmbed/mbed-os #6624: LocalFileSystem & wait() triggers MbedOS Fault Handler

yennster, 2018-04-12:
## Bug

**Target**
LPC1768

**Toolchain:**
GCC_ARM & Online Compiler

**Toolchain version:**
gcc-arm-none-eabi-7-2017-q4-major

**mbed-cli version:** 
1.5.0

**mbed-os sha:**
0b37629 (HEAD -> master, origin/master, origin/HEAD) Merge pull request #114 from adbridge/master

**Steps to reproduce**
I wrote up a simple `LocalFileSystem` example for the Mbed Application board that writes 100 integer values to a .csv file:
```
#include "mbed.h"
LocalFileSystem local("local");
int main() {
    FILE *fp = fopen("/local/test.csv","w");
    for (int i = 0; i < 100; i++) {
        fprintf(fp, "%d\n", i);
    }
    fclose(fp);
}
```
The above code works and a `TEST.CSV` file is created in the local filesystem with integer values 0-99. However, when I add a `wait` function to the `for` loop, the MbedOS Fault Handler is triggered:

```
#include "mbed.h"
LocalFileSystem local("local");
int main() {
    FILE *fp = fopen("/local/test.csv","w");
    for (int i = 0; i < 100; i++) {
        fprintf(fp, "%d\n", i);
        wait(0.05); // triggers ++ MbedOS Fault Handler ++
    }
    fclose(fp);
}
```
No values are stored into the csv file as the application doesn't get to `fclose(fp);`

**Expected behavior**
That the fault handler is not triggered and the for loop waits for 0.05 seconds, then saves a test.csv file.

**Actual behavior**
Output to the serial terminal of the Fault Handler error message:
```
                                        ++ MbedOS Fault Handler ++

                                                                  FaultType: HardFault

                                                                                      Context:
                                                                                              R0   : 00000005
                                                                                                             R1   : 100025AC
                                                                                                                            R2   : 00000122
                                                                                                                                           R3   : 00000000
          R4   : 00000005
                         R5   : 100025AC
                                        R6   : 10002920
                                                       R7   : 10000288
                                                                      R8   : 10000288
                                                                                     R9   : 00000000
                                                                                                    R10  : 00000000
                                                                                                                   R11  : 00000000
                                                                                                                                  R12  : 00002CD9
 SP   : 100025A8
                LR   : 00000CF1
                               PC   : 00001794
                                              xPSR : 01000000
                                                             PSP  : 10002588
                                                                            MSP  : 10007FB8
                                                                                           CPUID: 412FC230
                                                                                                          HFSR : 80000000
                                                                                                                         MMFSR: 00000000
                                                                                                                                        BFSR : 00000000
       UFSR : 00000000
                      DFSR : 00000002
                                     AFSR : 00000000
                                                    SHCSR: 00000000
                                                                   Mode : Thread
                                                                                Priv : Privileged
                                                                                                 Stack: PSP

                                                                                                           Thread Info:
                                                                                                                       Current:
                                                                                                                               State: 00000002 EntryFn: 00001CF1 Stack Size: 00001000 Mem: 10001648 SP: 100025D0
                                                                Next:
                                                                     State: 00000002 EntryFn: 00001CF1 Stack Size: 00001000 Mem: 10001648 SP: 100025D0
      Wait Threads:
                   State: 00000083 EntryFn: 00003A8D Stack Size: 00000300 Mem: 10000F78 SP: 10001210
                                                                                                    Delay Threads:
                                                                                                                  Idle Thread:
                                                                                                                              State: 00000001 EntryFn: 00001E79 Stack Size: 00000200 Mem: 10001278 SP: 10001420

                                                               -- MbedOS Fault Handler --
```

kjbracey-arm, 2018-04-13:
Sounds like it's probably the same root cause as https://github.com/ARMmbed/mbed-os/issues/6578 .

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-6050*

Carmeloning, 2019-01-18:
Have you configure it？
I meet something like this

adbridge, 2020-10-13:
We've updated our automation, I will fix the requirements .