ARMmbed/mbed-os #6631: DISCO-L072CZ-LRWAN1 should be marked as tiny target

janjongboom, 2018-04-13:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Related issue: https://github.com/ARMmbed/mbed-os-example-lorawan/issues/58
- Priority: Major

## Bug

**Target**
DISCO-L072CZ-LRWAN1

**Toolchain:**
All toolchains, but on ARM the problem is less visible. On GCC ARM this target is not usable.

**Toolchain version:**
GCC ARM 6

**mbed-os sha:**
Mbed OS 5.8.2

**Expected behavior**

The [DISCO-L072CZ-LRWAN1](https://os.mbed.com/platforms/ST-Discovery-LRWAN1/) has only 20K of SRAM. Running Mbed OS 5.8 in default configuration with the mbed-os-example-lorawan program (which is kind of the idea of a LoRa board) runs out of memory immediately in GCC ARM and has bytes to spare on ARM profile. Compiling with newlib-nano / uARM (without RTOS) brings memory usage back to only 7K, which leaves plenty of room for the user application. https://github.com/ARMmbed/mbed-os-example-lorawan/issues/58 is an example of a user which has this problem. For now I created [mbed-os-example-lorawan-minimal](https://github.com/janjongboom/mbed-os-example-lorawan-minimal) but that's not a good solution.

**Actual behavior**

At the very least this target should be marked as tiny (do we still have that?) to make it compile with uARM and newlib-nano. In addition I'd think that the RTOS should not be loaded here as running extra threads is impossible anyway with the current footprint.

cc @sg-

0xc0170, 2018-04-13:
Based on the numbers, this target should be probably just for single threaded applications? (default lib: small?). I checked and it is not true at the moment.

@ARMmbed/team-st-mcd Please review

jeromecoutant, 2018-04-13:
Yes, if you think that allowing only single thread application should be the default configuration, it is possible.

Setting small lib only affect GCC compilation, maybe it is better to add macro MBED_RTOS_SINGLE_THREAD ?

Setting small lib is also adding "--specs=nano.specs", this makes c_strings test failed.

janjongboom, 2018-04-14:
@jeromecoutant Wouldn't setting the target as tiny not also use uARM? So then only IAR is problematic.

0xc0170, 2018-04-16:
To summarize:

- set `default_lib` to `small`
- default toolchain uARM

kivaisan, 2018-04-16:
I don't think this is good for our LoRa tests. We need an additional thread in our LoRa greentea tests.

janjongboom, 2018-04-17:
@kivaisan, but how would you run greentea in an extra thread on this target? It does not fit anyway.

kivaisan, 2018-04-17:
@janjongboom ARM Compiler or IAR compiled image works.

0xc0170, 2018-04-17:
> I don't think this is good for our LoRa tests. We need an additional thread in our LoRa greentea tests.

Test can still overwrite it I hope - same for an application.
 A question is if its by default should be set to just single thread.

kivaisan, 2018-04-17:
> Test can still overwrite it I hope - same for an application (not completely certain)

Ok, if that's the case then we are ok. 

Btw. This issue also valid for target MTB_MURATA_ABZ. It is the almost the same HW as DISCO-L072CZ-LRWAN1.

janjongboom, 2018-04-18:
@0xc0170 In newlib-nano malloc is not thread safe I think, @sg- told me to not use RTOS whenever you use newlib-nano.

jeromecoutant, 2018-04-25:
Hi
What is your current opinion ?
Should we update the default DISCO L072 configuration ?

closed 2018-05-16 by 0xc0170