ARMmbed/mbed-os #6633: Avoid truncating .isr_table from custom linker section on LPC176X

0x6d61726b, 2018-04-13:
### Description
- Type: Bug | Question
- Priority: Major

---------------------------------------------------------------
## Bug / Question

**Target**
LPC176X

**Toolchain:**
GCC_ARM (GNUARM)

**Expected behavior**
Custom ".isr" section shall be present in output .hex file

**Actual behavior**
Custom ".isr" section is missing in output .hex file

**Steps to reproduce**
0. Compile and link the project with an unmodified .ld file from /targets/TARGET_NXP/TARGET_LPC176X/device/TOOLCHAIN_GCC_ARM/LPC1768.ld

1. Modify the linker file to place the ".isr_vector" into a custom ".isr" section (LPC1768_isr.ld)

2. Compile and link the project with modified LPC1768_isr.ld file

3. List and compare the section headers of both .elf files with: arm-eabi-readelf -S <name>.elf

   Relevant entry of unmodified sections:
   Section Headers:
     [Nr] Name              Type            Addr     Off    Size   ES Flg Lk Inf Al
     [ 0]                   NULL            00000000 000000 000000 00      0   0  0
     [ 1] .text             PROGBITS        00000000 010000 00f530 00  AX  0   0  8

   Relevant entry of modified sections:
   Section Headers:
     [Nr] Name              Type            Addr     Off    Size   ES Flg Lk Inf Al
     [ 0]                   NULL            00000000 000000 000000 00      0   0  0
     [ 1] .isr              PROGBITS        00000000 0202e4 0000cc 00      0   0  4
     [ 2] .text             PROGBITS        00002000 002000 00f460 00  AX  0   0  8
   .hex file content:
     Block  0  Starts at: 0x2000  Ends at: 0x11683 (Length: 0xF684=63108)
     00002000:  10 B5 05 4C  23 78 33 B9   04 4B 13 B1  04 48 AF F3  ...L#x3..K...H..
     00002010:  00 80 01 23  23 70 10 BD   E8 02 00 10  00 00 00 00  ...##p..........
     00002020:  5C 14 01 00  08 4B 10 B5   1B B1 08 49  08 48 AF F3  \....K.....I.H..
     00002030:  00 80 08 48  03 68 03 B9   10 BD 07 4B  00 2B FB D0  ...H.h.....K.+..
     00002040:  BD E8 10 40  18 47 00 BF   00 00 00 00  EC 02 00 10  ...@.G..........
     00002050:  5C 14 01 00  E4 02 00 10   00 00 00 00  15 4B 00 2B  \............K.+

4. Patch the startup file and assign the ".isr_vector" the section attribute "allocated" in file /targets/TARGET_NXP/TARGET_LPC176X/device/TOOLCHAIN_GCC_ARM/startup_LPC17xx.S

   63c63
   <     .section .isr_vector
   ---
   >     .section .isr_vector,"a"

5. List the section headers of the patched version with the modified LPC1768_isr.ld file

   Section Headers:
     [Nr] Name              Type            Addr     Off    Size   ES Flg Lk Inf Al
     [ 0]                   NULL            00000000 000000 000000 00      0   0  0
     [ 1] .isr              PROGBITS        00000000 010000 0000cc 00   A  0   0  4
     [ 2] .text             PROGBITS        00002000 012000 00f460 00  AX  0   0  8
   .hex file content:
     Block  0  Starts at: 0x   0  Ends at: 0x  CB (Length: 0xCC=204)
     Block  1  Starts at: 0x2000  Ends at: 0x11683 (Length: 0xF684=63108)
     00000000:  E0 7F 00 10  55 A6 00 00   85 A6 00 00  87 A6 00 00  . ..U...........
     00000010:  89 A6 00 00  8B A6 00 00   8D A6 00 00  00 00 00 00  ................
     00000020:  00 00 00 00  00 00 00 00   00 00 00 00  45 C5 00 00  ............E...
     00000030:  91 A6 00 00  00 00 00 00   AF C5 00 00  E5 C5 00 00  ................
     00000040:  97 A6 00 00  97 A6 00 00   97 A6 00 00  97 A6 00 00  ................
     00000050:  97 A6 00 00  97 A6 00 00   97 A6 00 00  97 A6 00 00  ................
     00000060:  97 A6 00 00  97 A6 00 00   97 A6 00 00  97 A6 00 00  ................
     00000070:  97 A6 00 00  97 A6 00 00   97 A6 00 00  97 A6 00 00  ................
     00000080:  97 A6 00 00  97 A6 00 00   97 A6 00 00  97 A6 00 00  ................
     00000090:  97 A6 00 00  97 A6 00 00   97 A6 00 00  97 A6 00 00  ................
     000000A0:  97 A6 00 00  97 A6 00 00   97 A6 00 00  97 A6 00 00  ................
     000000B0:  97 A6 00 00  97 A6 00 00   97 A6 00 00  97 A6 00 00  ................
     000000C0:  97 A6 00 00  97 A6 00 00   97 A6 00 00               ............    
     00002000:  10 B5 05 4C  23 78 33 B9   04 4B 13 B1  04 48 AF F3  ...L#x3..K...H..
     00002010:  00 80 01 23  23 70 10 BD   E8 02 00 10  00 00 00 00  ...##p..........
     00002020:  5C 14 01 00  08 4B 10 B5   1B B1 08 49  08 48 AF F3  \....K.....I.H..
     00002030:  00 80 08 48  03 68 03 B9   10 BD 07 4B  00 2B FB D0  ...H.h.....K.+..
     00002040:  BD E8 10 40  18 47 00 BF   00 00 00 00  EC 02 00 10  ...@.G..........
     00002050:  5C 14 01 00  E4 02 00 10   00 00 00 00  15 4B 00 2B  \............K.+

----------------------------------------------------------------
## Question

- Is there a way to define the "allocated" section attribute in the linker file?
- Is it a suitable way to extend mbed-os LPC17xx.S startup file to avoid this bug?

0x6d61726b, 2018-04-13:
Sorry for the bad formatting, but Firefox ESR is not supported any longer by github which seems to result in non-available java-script stuff

0xc0170, 2018-04-13:
I reformatted a bit. I don't think we do any magic there that would stop you from editing linker file + startup file.

> Custom ".isr" section is missing in output .hex file

 I might need some details to understand the problem (lot of numbers in the pasted data). `.isr` is the default one. `isr_vector` is the one you define in the linker file but does not get to the output file ?

0x6d61726b, 2018-04-13:
If you ask why the custom .isr section was introduced in the .ld file, the question is simple:
We want to use the small 4K flash sectors to store arbitrary data and not the large 32K sectors that are located at the end of the flash. Unfortunately those sectors are located at the beginning of the flash (sectors 0 to 15).

here is an excerpt of the modified linker file:

/* Linker script for mbed LPC1768 */

/* Linker script to configure memory regions. */
MEMORY
{
  ISR    (rx)  : ORIGIN = 0x00000000, LENGTH = 4K                 /* ISR table, CRP */
  FLASH  (rx)  : ORIGIN = 0x00002000, LENGTH = (512K - 8K)        /* generic flash flash */
  RAM    (rwx) : ORIGIN = 0x100000C8, LENGTH = (32K - 0xC8 - 32)  /* topmost 32 bytes used by IAP functions */

  USB_RAM(rwx) : ORIGIN = 0x2007C000, LENGTH = 16K
  ETH_RAM(rwx) : ORIGIN = 0x20080000, LENGTH = 16K
}
...
SECTIONS
{
    .isr :
    {
        KEEP(*(.isr_vector))
    } > ISR
       
    .text :
    {
        *(.text*)
        KEEP(*(.init))
        KEEP(*(.fini))
...

0x6d61726b, 2018-04-13:
Additional information can be found here:
https://sourceware.org/binutils/docs/as/Section.html

If other sections are placed into the ".isr" section like the ".init" section, the ".isr" section automatically gets marked with the flags "Flg" "AX" for "allocated" and "executable".

0xc0170, 2018-04-16:
> The ".isr_vector" is also present in the original LPC1768.ld file but in the .text section. I have created a new memory location "ISR" and just moved the ".isr_vector" into it. All remaining sections are kept in the ".text" section which is located in the memory "FLASH".

The issue resolved ?

0x6d61726b, 2018-04-16:
> The issue resolved ?
Not by default. It can be resolved, if the section is marked with the "a" (=allocated) flag in source code, as described in item #4 of the the initial post (```.section .isr_vector,"a"```).

0x6d61726b, 2018-04-17:
Thanks for the feedback.
I can do the PR, but please give me a few days since I am pretty busy right now.

0xc0170, 2018-08-01:
> I can do the PR, but please give me a few days since I am pretty busy right now.

@0x6d61726b Any update ?

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-6051*

maclobdell, 2019-01-15:
@0x6d61726b  Thanks for reporting this and offering to do a PR.  What is the status?  Have you made a PR?

0x6d61726b, 2019-05-30:
I have written a small tool that modified all files that did not yet contain the "allocated" flag for the .isr_vector section. Sorry for the delay. I have committed, but not pushed the changes (I did the changes now to the latest master branch).

closed 2019-10-31 by ciarmcom