ARMmbed/mbed-os #6641: Question about NVStore::set_max_keys(uint16_t num_keys)

soramame21, 2018-04-15:
### Description
- Type:  Question
- Priority:  Major
I have question about **NVStore::set_max_keys()** under following condition over TMPM46B (Tohisba). 

1. NVStore::set_max_keys is called
2. Then get_max_possible_keys() is called at line X ASSERT verification.
3. In get_max_possible_keys(), line Y, _size is sector size 32K, and _min_prog_size is page size 4K, so max_possible_keys = 7.
4. When return back NVStore::set_max_keys() line X, the num_keys to be set must be smaller than max_possible_keys, therefore the maximum acceptable num_keys must be 6.   

```
uint16_t NVStore::get_max_possible_keys()
{
;;
Y    size_t max_possible_keys = _size / align_up(sizeof(nvstore_record_header_t) * 2, _min_prog_size) - 1;
    return (uint16_t)std::min(max_possible_keys, (size_t) last_reserved_key);
}

void NVStore::set_max_keys(uint16_t num_keys)
{
X    MBED_ASSERT(num_keys < get_max_possible_keys());
    _max_keys = num_keys;
;;
}

```
## Question

**Q1. Why should max_possible_keys not be 8 at line Y? in case sector size is 32K and page size is 4K.**
**Q2. Why should num_keys be smaller then max_possible_keys without equal condition at line X?**

0xc0170, 2018-04-16:
@davidsaada Please review

davidsaada, 2018-04-16:
Good questions, first off all.
> Q1. Why should max_possible_keys not be 8 at line Y? in case sector size is 32K and page size is 4K.

Other than the data records, we also have a "master record" at address 0. This will hold one program unit, hence the `-1` at the end of the calculation.

> Q2. Why should num_keys be smaller then max_possible_keys without equal condition at line X?

You are right, a less-than-equal condition should be more suitable here.

I must add that the 4KB page size in this board (is this really the case?) is pretty nasty when it comes to NVStore. It makes the writes very slow, and causes an allocation of a 4KB buffer in RAM for these writes. This also means that if you have number of keys close to max_possible_keys, then it will keep performing the garbage collection process, which takes a lot of time (and may block communication). This is really not a very natural configuration for NVStore.

soramame21, 2018-04-17:
@davidsaada thanks for explanation. the page size of TMPM64B is fixed 4K byte. 
In case page size of target is big, should we allocate multiple sectors? or just return a error to tell the target is not suitable for nvstore feature?
Thanks confirmed to change check condition for Maximum key number in set_max_keys().

davidsaada, 2018-08-02:
< In case page size of target is big, should we allocate multiple sectors? or just return a error to tell the target is not suitable for nvstore feature?
Thanks confirmed to change check condition for Maximum key number in set_max_keys().

Sorry for the (very) late response. This really depends on how much you need NVStore and how much spare room you have for it, keeping in mind you also need the flash for the application. If the former is true, then I need to add a validation to NVStore init, to check whether the maximal number of keys obeys to a minimum.