ARMmbed/mbed-os #6643: features-nvstore-tests-nvstore-functionality failed when can not accept max_test_keys

soramame21, 2018-04-16:
### Description
- Type: Enhancement
- Priority: Minor
TMPM46B failed features-nvstore-tests-nvstore-functionality. features\nvstore\TESTS\nvstore\functionality\main.cpp
Because the chip has fixed sector size 32K and page size 4K, therefore the maximum acceptable key number is 7, but our test case set the key number is 20.

## Question

**Q1: Why the test case set the key number 20? is it the minimum number to support?**

**Q2: In the test case, can we set the key number with value of maximum acceptable key number instead of a fixed number?**

0xc0170, 2018-04-16:
@davidsaada Please review

davidsaada, 2018-04-16:
Will try to answer both questions with one answer.
20 is not not the minimum number, but it's very close to the minimum.  Actually we aimed for NVStore to have 16 fixed keys and the rest to be dynamically allocated according to need (added a feature for it in 5.9). A configuration such as TMPM46B has only allows 7 keys (see [#6641](https://github.com/ARMmbed/mbed-os/issues/6641#issuecomment-381499271)), which is something we couldn't imagine.

cmonr, 2018-08-02:
Closing since it seems like the questions were answered.

closed 2018-08-02 by cmonr