ARMmbed/mbed-os #6649: LoRaWAN should have API for SNR and RSSI

janjongboom, 2018-04-17:
### Description
- Type:  Enhancement 
- Related issue: #6389
- Priority: Minor

## Enhancement

**Reason to enhance or problem with existing solution**

See also #6389. If you receive a message from the LoRaWAN stack it'd be useful to see RSSI and SNR (and probably some other stuff, f.e. was this a confirmed message?) of that packet for debugging purposes.

**Suggested enhancement**

Change the receive function so it takes in a struct, and put the information on there; instead of the out parameters that we use now.

**Pros**

Get more debug information.

**Cons**

Change the API again (third time in three releases :-)).

@kivaisan @hasnainvirk

kivaisan, 2018-04-18:
Isn't adding members to such a "collector" structure in a public interface be a API break?
@AnttiKauppila what do you think?

hasnainvirk, 2018-04-18:
In my humble opinion we shouldn't touch the receive APIs anymore. If metadata is needed, it should be retrieved by some out of band means. Let's not bloat out our APIs

hasnainvirk, 2018-06-27:
https://github.com/ARMmbed/mbed-os/blob/master/features/lorawan/LoRaWANInterface.h#L469 meta data APIs were added to the LoRaWANInterface as per 5.9. I hope that solves the issue.

0xc0170, 2018-06-28:
> https://github.com/ARMmbed/mbed-os/blob/master/features/lorawan/LoRaWANInterface.h#L469 meta data APIs were added to the LoRaWANInterface as per 5.9. I hope that solves the issue.

Please reopen if this does not fixes the issue or any other action needed here

closed 2018-06-28 by 0xc0170

janjongboom, 2018-07-09:
@hasnainvirk Nice.