ARMmbed/mbed-os #6650: Greentea UDP/TCP packet pressure test cases are failing on platforms with large memory

mikaleppanen, 2018-04-17:
### Description
- Type: Bug
- Related issue: None
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_F767ZI

**Toolchain:**
GCC_ARM

**Toolchain version:**
6-2017-q2

**Expected behavior**

Greentea UDP and TCP packet pressure tests should pass:

    mbed-os-tests-netsocket-tcp_packet_pressure
    mbed-os-tests-netsocket-tcp_packet_pressure_parallel
    mbed-os-tests-netsocket-udp_packet_pressure
    mbed-os-tests-netsocket-udp_packet_pressure_parallel

**Actual behavior**

Greentea UDP and TCP packet pressure tests fail on platforms with large memories. 

This is due to overflow on lwip socket send/receive where application socket buffer length is cast from uint32 to uint16 without checking if value is larger than uint16. 

Also, in case platform has more memory than configured maximum, test case generate_buffer() routine does not divide the reserved memory by four, before trying to allocate it.

Fixes needed to both lwip and test cases.

**Steps to reproduce**

Run tests on platform with large memory (more than >128kbytes)

jeromecoutant, 2018-05-16:
Hi

Any update ?

Thx

mikaleppanen, 2018-05-17:
Test will be replaced with new net socket tests in: https://github.com/ARMmbed/mbed-os/pull/6665

There will not be updates to old ones, but the new tests has to be verified with these boards (e.g. NUCLEO_F767ZI) to ensure that there are no new test case problems.

jeromecoutant, 2018-05-30:
Maybe this issue can be closed ?

mikaleppanen, 2018-06-15:
New tests have been released in 5.9. This is no longer valid. Closing this.

closed 2018-06-15 by mikaleppanen