ARMmbed/mbed-os #6662: How to get RTC value in ms. precision?

janjongboom, 2018-04-18:
### Description
- Type:  Question

## Question

**How to?**

I need to know how long my application has been running. `gettimeofday()` is available but always returns `0` for `tv_sec` and `tv_usec`. `time(NULL)` just gives me second precision. Using a timer is not recommended because it overflows after 31 minutes... How can I get the RTC value with ms. precision?

For Mbed OS 3 there was [mbed-time](https://github.com/ARMmbed/mbed-time/) but it requires lp ticker and the overflow counter function is not available in Mbed OS 5.

0xc0170, 2018-04-18:
@SenRamakri @bulislaw Please review

kjbracey-arm, 2018-04-19:
In the C library, I'd hope `clock()` works, but I'm not certain.

mbed OS 5.8 added `Kernel::get_ms_count()`, which gets the millisecond tick counter from the RTOS, expanding to 64 bits if necessary. That's what I'd recommend as the best general-purpose timebase, and a number of delaying RTOS calls now accept that 64-bit absolute time as an "until" value.

kjbracey-arm, 2018-04-19:
This was pending work from the discussion on `Kernel::get_ms_count()` - what should the non-RTOS version be, and could it be interfaced into `clock()`?

Main issue with `clock()` is we should ideally be able to get control of `CLOCKS_PER_SEC` to make sure it's 1000 to reflect reality for RTOS and align non-RTOS - I'd guess each C library should have a mechanism for this, but we're currently converting microseconds to whatever the library defaults to, I think (centiseconds?) Changing historic behaviour is a potential compatibility break, of course.

kjbracey-arm, 2018-04-20:
For the high-res stuff, there is the misleadingly named `Timer::read_high_resolution_us()`. That returns the same thing as `Timer::read_us()`, but it's 64-bit, so shouldn't warp for some millenia.

Internal implementation of `Timer` is 64-bit - it appears `read_us` is 32-bit just for backwards compatibility.

janjongboom, 2018-04-22:
@kjbracey-arm Awesome, thanks.

closed 2018-04-22 by janjongboom