ARMmbed/mbed-os #6670: Macro definitions in mbed_app.json are not passed to IAR assembler

TeroJaasko, 2018-04-18:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug 
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
IAR

**Toolchain version:**
8.2

**mbed-cli version:** 
1.5.0

**mbed-os sha:**
cba28cc (HEAD -> master, origin/master, origin/HEAD) Merge pull request #6221 from codeauroraforum/Add_RNG_LPC54XXX

**Expected behavior**
Code compiles, macros given in mbed_app.json are passed to assembler.

**Actual behavior**
```
Building project mbed-os-example-blinky (K64F, IAR)
Scan: .
Scan: mbed
Scan: env
Scan: FEATURE_LWIP
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_STORAGE
Link: mbed-os-example-blinky
Error[Li005]: no definition for "mbed_fault_context" [referenced from
          mbed-os-example-blinky/BUILD/K64F/IAR/mbed-os/rtos/TARGET_CORTEX/TARGET_CORTEX_M/TOOLCHAIN_IAR/except.o]
Error[Li005]: no definition for "mbed_fault_handler" [referenced from
          mbed-os-example-blinky/BUILD/K64F/IAR/mbed-os/rtos/TARGET_CORTEX/TARGET_CORTEX_M/TOOLCHAIN_IAR/except.o]
[ERROR] Error[Li005]: no definition for "mbed_fault_context" [referenced from
          mbed-os-example-blinky/BUILD/K64F/IAR/mbed-os/rtos/TARGET_CORTEX/TARGET_CORTEX_M/TOOLCHAIN_IAR/except.o]
Error[Li005]: no definition for "mbed_fault_handler" [referenced from
          mbed-os-example-blinky/BUILD/K64F/IAR/mbed-os/rtos/TARGET_CORTEX/TARGET_CORTEX_M/TOOLCHAIN_IAR/except.o]

[mbed] ERROR: "/usr/bin/python" returned error code 1.
[mbed] ERROR: Command "/usr/bin/python -u mbed-os-example-blinky/mbed-os/tools/make.py -t IAR -m K64F --source . --build ./BUILD/K64F/IAR" in "mbed-os-example-blinky"
```

**Steps to reproduce**
* clone mbed-os-example-blinky
* compile successfully for ARM: "mbed compile -m K64F -t ARM"
* edit mbed_app.json to have macro MBED_FAULT_HANDLER_DISABLED as defined, eg
```
    "macros": [
        "MBED_FAULT_HANDLER_DISABLED"
    ],
```
* compile successfully for ARM: "mbed compile -m K64F -t ARM"
* observe binary size reduction of ~1780 bytes.
* compile for IAR: "mbed compile -m K64F -t IAR"
* observe the linker error, caused likely by "#ifndef MBED_FAULT_HANDLER_DISABLED" -line failing to work as expected in except.S (https://github.com/ARMmbed/mbed-os/blob/master/rtos/TARGET_CORTEX/TARGET_CORTEX_M/TOOLCHAIN_IAR/except.S#L32)

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**
It seems that the macros in mbed_app.json are not passed to the IAR's assembler, or at least command lines for compiling except.S in ARM and IAR are quite different and IAR is missing reference to mbed_config.h:

ARM:
```
armcc -c --gnu -Otime --split_sections --apcs=interwork --brief_diagnostics --restrict --multibyte_chars -O3 --cpu=Cortex-M4.fp -D__CORTEX_M4 -DCPU_MK64FN1M0VMD12 -DARM_MATH_CM4 -D__FPU_PRESENT=1 -D__MBED_CMSIS_RTOS_CM -D__CMSIS_RTOS -DFSL_RTOS_MBED --via ./BUILD/K64F/ARM/.includes_24c7dc53f90030b3047073a06c584023.txt --preinclude=./BUILD/K64F/ARM/mbed_config.h -E -o BUILD/K64F/ARM/mbed-os/rtos/TARGET_CORTEX/TARGET_CORTEX_M/TOOLCHAIN_ARM/.temp/except.o.E.s ./mbed-os/rtos/TARGET_CORTEX/TARGET_CORTEX_M/TOOLCHAIN_ARM/except.S
```

IAR:
```
iasmarm --cpu Cortex-M4F -D__CORTEX_M4 -DCPU_MK64FN1M0VMD12 -DARM_MATH_CM4 -D__FPU_PRESENT=1 -D__MBED_CMSIS_RTOS_CM -D__CMSIS_RTOS -DFSL_RTOS_MBED -o BUILD/K64F/IAR/mbed-os/rtos/TARGET_CORTEX/TARGET_CORTEX_M/TOOLCHAIN_IAR/except.o ./mbed-os/rtos/TARGET_CORTEX/TARGET_CORTEX_M/TOOLCHAIN_IAR/except.S
```

**Suggested enhancement**
Pass the content of mbed_config.h to iasmarm somehow.

**Pros**

**Cons**

-----------------------------------------------------------------

## Question

**How to?**

theotherjimmy, 2018-05-08:
@TeroJaasko When did you try the arm assembler last? I get the following:
```bash
armcc -c --gnu -Otime --split_sections --apcs=interwork --brief_diagnostics --restrict --multibyte_chars -O3 --cpu=Cortex-M4.fp -D__CORTEX_M4 -DARM_MATH_CM4 -D__FPU_PRESENT=1 -D__MBED_CMSIS_RTOS_CM -D__CMSIS_RTOS -DFSL_RTOS_MBED -DCPU_MK66FN2M0VMD18 -E -o BUILD/k66f/arm/mbed-os/rtos/TARGET_CORTEX/TARGET_CORTEX_M/TOOLCHAIN_ARM/.temp/except.o.E.s ./mbed-os/rtos/TARGET_CORTEX/TARGET_CORTEX_M/TOOLCHAIN_ARM/except.S
```

Which does not include any mention of mbed_config.h

TeroJaasko, 2018-05-09:
@theotherjimmy this seems to be now broken in Mbed OS master with ARMCC also. 

mbed OS cba28cc0ac2, the mbed_config.h  is given to armcc:
```
armcc -c --gnu -Ospace --split_sections --apcs=interwork --brief_diagnostics --restrict --multibyte_chars -O3 -DNDEBUG --cpu=Cortex-M4.fp -D__CORTEX_M4 -DCPU_MK64FN1M0VMD12 -DARM_MATH_CM4 -D__FPU_PRESENT=1 -D__MBED_CMSIS_RTOS_CM -D__CMSIS_RTOS -DFSL_RTOS_MBED --via ./BUILD/K64F/ARM/.includes_24c7dc53f90030b3047073a06c584023.txt --preinclude=./BUILD/K64F/ARM/mbed_config.h -E -o BUILD/K64F/ARM/mbed-os/rtos/TARGET_CORTEX/TARGET_CORTEX_M/TOOLCHAIN_ARM/.temp/except.o.E.s ./mbed-os/rtos/TARGET_CORTEX/TARGET_CORTEX_M/TOOLCHAIN_ARM/except.S
```

Mbed OS 68ad00ffb8, no mbed_config.h given anymore:
```
armcc -c --gnu -Ospace --split_sections --apcs=interwork --brief_diagnostics --restrict --multibyte_chars -O3 -DNDEBUG --cpu=Cortex-M4.fp -D__CORTEX_M4 -DCPU_MK64FN1M0VMD12 -DARM_MATH_CM4 -D__FPU_PRESENT=1 -D__MBED_CMSIS_RTOS_CM -D__CMSIS_RTOS -DFSL_RTOS_MBED -E -o BUILD/K64F/ARM/mbed-os/rtos/TARGET_CORTEX/TARGET_CORTEX_M/TOOLCHAIN_ARM/.temp/except.o.E.s ./mbed-os/rtos/TARGET_CORTEX/TARGET_CORTEX_M/TOOLCHAIN_ARM/except.S
```

theotherjimmy, 2018-05-09:
@TeroJaasko It looks like GCC doing it is a bug, where the implementer forgot to say "this is for assembly" when invoking the asm preprocessor.

theotherjimmy, 2018-06-01:
@TeroJaasko I'm pretty sure that assembly cannot depend on any configuration system values as it stands. This is now a feature request to extend the asm command line to include macros defined in the configuration system.

commit dcd17935 (2018-07-20): IAR: Fixed #6670
commit 91d8d0c7 (2018-07-27): IAR: Fixed #6670
commit 71ea0489 (2018-08-10): IAR: Fixed #6670
closed 2018-07-20 by cmonr