ARMmbed/mbed-os #6674: Greentea client isn't outputting to the console

kjbracey-arm, 2018-04-19:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Enhancement 
- Related issue: #6660
- Priority: Minor

----------------------------------------------------------------
## Enhancement

Greentea client is unnecessarily tied to `RawSerial`, potentially limiting performance, and preventing its use over any other comms interface. It's also causing configuration complications with tracking the serial port settings specified for the stdout console, as illustrated by #6660.

It's apparently avoiding the simplest `printf` and `putc` to permit use from interrupt (https://github.com/ARMmbed/greentea-client/commit/ff623719a14c2f22cad8d55fba815ebeac4352b4).

This can be now be achieved in a more elegant and portable fashion using `write(STDOUT_FILENO)` - this will in principle work from interrupt, subject to limitations of the `FileHandle` being used for the console. It will currently work with the default non-buffered console, equivalent to `RawSerial`.

A little more discussion in comments on #6660

0xc0170, 2018-04-19:
@SenRamakri @bulislaw Please review

> It's apparently avoiding the simplest printf and putc to permit use from interrupt (ARMmbed/greentea-client@ff62371).

cc @adbridge I checked the original addition to greentea to start using RawSerial  done by you so might be interested in this.

kjbracey-arm, 2018-04-19:
I am curious why you're trying to output from interrupt context. Yes, you can use `RawSerial` or `write(STDOUT_FILENO)` to do that, but doesn't mean you should.

I'm advocating use of `write(STDERR_FILENO)` to get fault, assert and other "die" handlers to send stuff to arbitrary console, removing their serial dependency (should probably raise an issue to track that if there isn't one).

But holding up a _running_ system in IRQ context to wait for the serial port is a different issue. More than 1 character at 115200 blows the 100us IRQ latency that is usually a good guideline.  I guess as long as it's just test code, or the IRQ messages are teeny?

theotherjimmy, 2018-04-19:
@adbridge This is not under my remit. I maintain the greentea host side of things.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-336*

closed 2019-10-31 by ciarmcom