ARMmbed/mbed-os #6685: Uninitialized variable in flashJournalStrategySequential_read_progress

markhermelinggt, 2018-04-19:
In flashJournalStrategySequential_read_progress in support_funcs.c

Line 349, if journal->read.amountLeftToRead is false, then the rest of that condition is not executed and hence storageBlock is not initialized when it is used in line 354.

Found during static analysis using CodeSonar, screenshot attached

[image]

Likely best fix would be to storageBlockAvailableCapacity to 0 in this case?

0xc0170, 2018-04-20:
Thanks for the report. Can you send a pull request fixing it?

Just a note: flashjournal was deprecated since Mbed OS 5.5

markhermelinggt, 2018-04-28:
Martin,

I am happy to send a fix and pull request, but unfortunately, I am not qualified to test it.

As this is deprecated, maybe the best would be to mark this as a wont fix?

Regards,
Mark

—
Mark Hermeling | GrammaTech | Senior Director Product Marketing
mobile +1 (607) 351-5719 | http://www.grammatech <http://www.grammatech/>.com/

New: Visit http://community.grammatech.com/ <http://community.grammatech.com/> for CodeSonar tips and tricks!

0xc0170, 2018-05-22:
Marked as won't fix as mentioned above.

closed 2018-05-22 by 0xc0170