ARMmbed/mbed-os #6694: How to generate a true random sequence

davidsaada, 2018-04-20:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Question
- Related issue: TRNG
- Priority: Major

## Question

**How to?** 
Would like to know the preferred way to generate a true random sequence.
StorageLite code needs to generate an 8 byte random nonce for each stored file. Now, assuming the device has a TRNG, we should use it for this purpose. Using HAL APIs from the feature code doesn't seem to be the proper way, thus using APIs in `trng_api.h` seems wrong. It seems like the correct way is to use the `mbedtls_hardware_poll` API. So, looks like my code should copy and paste the `mbedtls_entropy_source_self_test_gather` static (test?) function in `entropy.c` file. But not sure about it. So my questions are:
1. Is this the right way to generate a true random sequence?
2. What should we do if we don't have a TRNG (`srand` + `rand`)?
3. If assumption regarding the usage of TRNG is correct, why not export the `mbedtls_entropy_source_self_test_gather` function (or an equivalent one)? Seems like this is the most convenient way to work here.

0xc0170, 2018-04-20:
@hanno-arm Please review

hanno-arm, 2018-04-20:
Hi @davidsaada,

the preferred way to get entropy is through Mbed TLS' entropy module, which recognizes, queries and mixes the entropy sources available on your platform. In particular, if a platform-specific TRNG driver has been implemented, this is called from `mbedtls_hardware_poll`, which in turn is registered as one of the entropy sources used by the entropy module.

The example program [gen_entropy](https://github.com/ARMmbed/mbedtls/blob/development/programs/random/gen_entropy.c) in the Mbed TLS repository exemplifies how to use it.

Please let me know if you have further questions.

Kind regards,
Hanno, Mbed TLS team member

davidsaada, 2018-04-20:
Thanks a lot. The example is very clear. Just one question: Can an application have more than one entropy context in parallel? Aim is to have such a context as a private class member, regardless of ones that may be used by the app.

hanno-arm, 2018-04-20:
Hi @davidsaada,

If you enable `MBEDTLS_THREADING_C`, then according to the [documentation of `mbedtls_entropy_func`](https://github.com/ARMmbed/mbedtls/blob/development/include/mbedtls/entropy.h#L193) you are fine. If you cannot enable this option for some reason, you need to take care of the mutexing yourself.

Could you give more information on your intended simultaenous use across multiple applications? Note also that if you require large amounts of random data, you should consider coupling the entropy module with the a CTR DRBG pseudo-RNG instead of directly acquiring it from the entropy context.

Kind regards,
Hanno

davidsaada, 2018-04-20:
@hanno-arm
I don't need simultanoeus access to entropy in my class. My class has a mutex that lets a single thread use the entropy module at a time (among the rest). Reason I asked was that I would like to allow the application to use entropy regardless of my class. I also don't need to generate a lot of random data - only 8 bytes at a time.

hanno-arm, 2018-04-20:
Hi @davidsaada,

ok, let me know if you have further questions then!

Kind regards,
Hanno

davidsaada, 2018-04-20:
No further questions. Thanks a lot for your help.

closed 2018-04-20 by davidsaada