ARMmbed/mbed-os #6695: NanostackRfPhyEfr32 driver does not build

SeppoTakalo, 2018-04-20:
### Description
- Type: Bug

`NanostackRfPhyEfr32` references `RAIL_IEEE802154_Init()` function that is introduced in `targets/TARGET_Silicon_Labs/TARGET_SL_RAIL/efr32-rf-driver/rail/ieee802154/rail_ieee802154.h` header. However, no implementation of it exists in build tree.

This `SL_RAIL` label was enabled on following MCUs:
* EFR32MG1P132F256GM48
* EFR32MG1P233F256GM48
* EFR32MG12P332F1024GL125

Needs fix from Silicon Labs.

SeppoTakalo, 2018-04-20:
CC: @stevew817 @0xc0170

0xc0170, 2018-04-20:
There's librail library in the tree, but the question is - what compiler was used to generate (from one earlier PR there was a mention gcc, also name indicates that)? 

Our CI failed to link with ARMCC. Any binary lib should have all 3 compilers supported.

stevew817, 2018-04-23:
Should be fixed by PR #6701

closed 2018-05-22 by 0xc0170